rusefi/rusefi #7019: Idle Solenoid Output mode - inverted output pulldown when engine not running

Tomika012345, 2024-10-31:
If inverted Idle Solenoid output mode is selected, corresponding pin is always low when engine is not running. This grounds the idle valve negative, applying max load on it which is unnecessary. 

The idle pwm output should not be pulled down at 0 rpm regardless of output mode selected. 

Thank You
[image]

rusefillc, 2024-10-31:
@Tomika012345 please help my explicitly stating what board are you using?

@Tomika012345 how do you understand the meaning of "default inverted" output mode? I wonder if we have a confusion/usability issue here

ElDominio, 2024-10-31:
So this issue boils down to two different things, I will make a new issue about part 2;

1) (current ticket) MOSFET control settings should be hidden on known boards. Changing these settings changes how electrical stuff works, these things need to be hidden from the user.

Tomika012345, 2024-10-31:
@rusefillc  - board is uaefi standard. Default inverted is the selected option in TS as seen in the screenshot. My current use for it is a vw/seat/skoda throttle body, without any input the throttle is stopped at partially open position, cca 5%. Throttle cable opens the throttle blade more, pwm operated idle motor pulls down the stopper of the throttle blade so it closes more. So i need to inversely apply pwm, "default inverted" mode works good. The only problem is that when put ignition on before cranking, the idle output gives constant earth to the motor until any rpm is registered. If i select "default" output mode, the idle output is left floating, so the idle motor is not moved at all until rpm is registered.