rusefi/rusefi #7026: Closed loop fuel correction I-term grows infinitely

ElDominio, 2024-11-02:
**Describe the bug**
i-term seems to be growing when stuck at one of the limits, as shown at 3.7s on the attached log. Correction is stuck at -5 for a while (while our target is 13.2 and actual is 17, definitely we do NOT want fuel to be removed) until it eventually starts adding fuel again. i-term should stop updating when we hit the correction limits (since we want to correct instantly out of the bounds once conditions are correct)

**Hardware**
 - Board: any

**To Reproduce**
Steps to reproduce the behavior:
1. Turn on closed loop fuel
2. allow Fuel Trim to be stuck at a bound (force fuel to be 2 afr under target with  correction at negative maximum)
3. change fuel just enough to be slightly under said bound
4. watch fuel trim take a long time to react

**Expected behavior**
Fuel trim should update the moment we need additional/less fuel when lying on a bound (pause all PID when maximum/minimum correction is reached)

**Additional context**
https://rusefi.com/online/view.php?log=1433

FixyNeko, 2024-11-19:
I term limit should be baked into PID logic, linked with #6900

mck1117, 2024-11-19:
> I term limit should be baked into PID logic, linked with #6900

closed loop fuel is not pid