rusefi/rusefi #7030: How does the ECU decide when to use MAP estimate?

ElDominio, 2024-11-02:
**Describe the bug**
It is not clear when the ECU uses the MAP estimate table. Are there thresholds to use it? Are they user-configurable?

**To Reproduce**
Steps to reproduce the behavior:
1. Have a MAP that reads 0.1v
2. MAP shown is 0kpa
3. Should be using MAP estimate from estimate table

**Expected behavior**
When MAP is considered failed (who decides this?), we should fallback to MAP estimate when running Speed Density.

FDSoftware, 2024-11-03:
**TLDR**: rusEFI (for now) does not decide in any (practical) case the use of the MAP estimate table

nerd research below:

checking:
`firmware/controllers/algo/airmass/speed_density_airmass.cpp`
```c++
float SpeedDensityAirmass::getMap(float rpm, bool postState) const {
	float fallbackMap = m_mapEstimationTable->getValue(rpm, Sensor::getOrZero(SensorType::Tps1));

#if EFI_TUNER_STUDIO
	if (postState) {
		engine->outputChannels.fallbackMap = fallbackMap;
	}
#endif // EFI_TUNER_STUDIO

	return Sensor::get(SensorType::Map).value_or(fallbackMap);
}

``` 
and:
`firmware/controllers/engine_cycle/map_averaging.cpp`
```c++
#if HAL_USE_ADC

/**
 * This method is invoked from ADC callback.
 * @note This method is invoked OFTEN, this method is a potential bottleneck - the implementation should be
 * as fast as possible
 */
void mapAveragingAdcCallback(float instantVoltage) {
	efiAssertVoid(ObdCode::CUSTOM_ERR_6650, hasLotsOfRemainingStack(), "lowstck#9a");

	SensorResult mapResult = getMapAvg(currentMapAverager).submit(instantVoltage);

	if (!mapResult) {
		// hopefully this warning is not too much CPU consumption for fast ADC callback
		warning(ObdCode::CUSTOM_INSTANT_MAP_DECODING, "Invalid MAP at %f", instantVoltage);
	}

#if EFI_TUNER_STUDIO
	float instantMap = mapResult.value_or(0);
	engine->outputChannels.instantMAPValue = instantMap;
#endif // EFI_TUNER_STUDIO
}
#endif
``` 
and:
`firmware/init/sensor/init_map.cpp`
```c++
// lowest reasonable idle is maybe 600 rpm
// one sample per cycle (1 cylinder, or "sample one cyl" mode) gives a period of 100ms
// add some margin -> 200ms timeout for fast MAP sampling
MapAverager fastMapSensor(SensorType::MapFast, MS2NT(200));
MapAverager fastMapSensor2(SensorType::MapFast2, MS2NT(200));

MapAverager& getMapAvg(size_t idx) {
	return idx == 0 ? fastMapSensor : fastMapSensor2;
}

``` 
and (finally):
`firmware/controllers/engine_cycle/map_averaging.cpp`

```c++
SensorResult MapAverager::submit(float volts) {
	auto result = m_function ? m_function->convert(volts) : unexpected;

	if (m_isAveraging && result) {
		chibios_rt::CriticalSectionLocker csl;

		m_counter++;
		m_sum += result.Value;
	}

	return result;
}

``` 
The estimated map will be used in cases where the sensor is not initialized, since the low/high voltages are not checked throughout the implementation, although they are defined in other parts of the code. I think the value of 0Kpa in TunerStudio is due to this section:

```c++
#if EFI_TUNER_STUDIO
	float instantMap = mapResult.value_or(0);
	engine->outputChannels.instantMAPValue = instantMap;
#endif // EFI_TUNER_STUDIO
```

rusefillc, 2024-11-03:
Invalid MAP detection could be related to the below

[image]

ElDominio, 2024-11-03:
> Invalid MAP detection could be related to the below
> 
> [image]
> 

It used to be? I know I have used that successfully but I could not get it to work today

rusefillc, 2024-11-03:
@ElDominio just added new logging data point / trying to add explicit indicator right on that dialog https://github.com/rusefi/rusefi/commit/ed8fbde6e7b722579e951e8c3186db595c45f76f

rusefillc, 2024-11-03:
@ElDominio colors are odd but text seems to change based on thresholds

[image]

[image]

closed 2024-11-03 by rusefillc

ElDominio, 2024-11-09:
Change "happy" to "good" for professional reasons