rusefi/rusefi #7034: Real hardware ICU coverage on nucleo-F767

rusefillc, 2024-11-03:
Once we have https://github.com/rusefi/rusefi/issues/7033

Same as discovery HW CI uses two jumper wires for continues integration of real pin toggle and real ICU peripheral we should add similar two jumpers on the nucleo 767

https://github.com/rusefi/rusefi/wiki/Dev-Quality-Control#stm32f407g-disc1

rusefillc, 2024-12-04:
@FDSoftware can you help?

rusefillc, 2024-12-18:
@FDSoftware we have ICU what about

```
//            MiscTest.class,
//            BurnCommandTest.class,
//            CommonFunctionalTest.class,
```

which one of these three is simplest to activate?

FDSoftware, 2024-12-18:
`MiscTest` It seems not, actual temp on the F7 is 25°C, i got 3° from `getMCUInternalTemperature`
 `BurnCommandTest` seems to be ideal, now I upload PR, It works without requiring any changes to the fw
log:  https://gist.github.com/FDSoftware/a33cc087f37b0b77fbad27cfeb02e289