rusefi/rusefi #7048: External aem wideband does not work over CAN

ElDominio, 2024-11-06:
I set the onboard wideband to index 1, and after reboot, it seems to set itself to index 0, showing overlapping wideband statuses in the gauge[image]

rusefillc, 2024-11-06:
@ElDominio I've added "Suppress on-board WBO" workaround but real issue is that WBO firmware does not persist index :(

@mck1117 @dron0gus

ElDominio, 2025-01-06:
@mck1117 had there been progress on this

EA11R, 2025-02-19:
I've came up with the idea to create two update firmware buttons , one that actually loads the FW for the wideband,  one that loads a FW with the CAN BUS broadcast disabled and keeping everything else intact 

I did figure the parts I need to change in WBO firmware but I need someone to point me to what should I do in the tuner studio part to make it work with UAEFI boards automatically rather than manually

ElDominio, 2025-02-19:
I don't understand the last question, but the two firmware update buttons sound great. Maybe label them (Flash firmware, flash firmware with CAN disabled)

dron0gus, 2025-05-12:
https://github.com/mck1117/wideband/blob/5c20f128e86bdc9c5d7b5a182f3ab10287dee4f9/firmware/boards/f0_module/port.cpp#L128
FW picks ID from saved settings only if both ID pins are floating...
This is not true for RE boards:

[image]

@rusefillc I'm going to make FW read ID pins only in case of no valid settings in flash.

dron0gus, 2025-05-19:
Part one https://github.com/rusefi/rusefi/pull/7917

dron0gus, 2025-05-19:
@ElDominio can you test?

rusefillc, 2025-06-29:
@ElDominio can you test?