rusefi/rusefi #7054: Allow up to 510 as Y axis in boost tables that use MAP as Y axis

ElDominio, 2024-11-08:
**Is your feature request related to a problem? Please describe.**
I want to use 500 as a Y value in a boost table that uses MAP as Y. Currently max is 255. I understand 510 is possible by multiplying *2

**Would we need new configuration fields?**
None

**Describe the solution you'd like**
Multiply Y axis by 2

**Describe alternatives you've considered**
None really

rusefillc, 2024-11-08:
```
! technical debt: some of these use uint8_t !!!
#define MAP_UPPER_LIMIT 255
```

[image]

FDSoftware, 2024-11-13:
@ElDominio you can try now this with the build: [Nightly 2024-11-13](https://github.com/rusefi/rusefi/releases/tag/2024-11-13)

ElDominio, 2025-01-18:
@rusefillc this seems fixed?

closed 2025-01-18 by rusefillc