rusefi/rusefi #7076: TIM1_CH4 ICU

rusefillc, 2024-11-15:
What's the level of effort to support ICU on TIM1_CH4?

rusefillc, 2024-11-18:
требуется минимальный рабочий вариант, чтоб заработал конкретно TIM1 CH4 на F7

engineConfiguration->sentInputPins[0] = Gpio::H144_IN_AUX2_DIGITAL;

минимальный тест - руками toggle пина, инкремент счётчика. см. текущие коммиты по тикету

rusefillc, 2024-11-19:
@andreika-git т.е. заработать должно только https://github.com/rusefi/rusefi/commit/004e02fa0dbf00a3bae1ff86e6e7262a2429e74b

rusefillc, 2024-11-19:
@andreika-git 8chan? nucleo F7?

rusefillc, 2024-12-02:
@dron0gus while some activity seems to be happening my test throttle body is not being decoded. same ETB is decoded by uaEFI just fine.

```
2024-12-01_21_48_12_479: EngineState: --------------------
2024-12-01_21_48_12_480: EngineState: confirmation_sentinfo:8
2024-12-01_21_48_12_765: EngineState: ---- SENT ch 0 ---- on D5 FLEX
2024-12-01_21_48_12_768: EngineState: Unit time 539 CPU ticks 3.208333248 uS
2024-12-01_21_48_12_768: EngineState: Total pulses 402398
2024-12-01_21_48_12_769: EngineState: Last valid fast msg Status 0x0 Sig0 0xAD2 Sig1 0x52D
2024-12-01_21_48_12_770: EngineState: Restarts 0
2024-12-01_21_48_12_770: EngineState: Interval errors 4235 short, 474 long
2024-12-01_21_48_12_775: EngineState: Total frames 39703 with CRC error 45 (0.113341560%)
2024-12-01_21_48_12_776: EngineState: Total slow channel messages 0 with crc6 errors 0 (NaN%)
2024-12-01_21_48_12_787: EngineState: Sync errors 4235
2024-12-01_21_48_12_788: EngineState: --------------------
2024-12-01_21_48_12_790: EngineState: confirmation_sentinfo:8
2024-12-01_21_48_12_865: EngineState: ---- SENT ch 0 ---- on D5 FLEX
2024-12-01_21_48_12_865: EngineState: Unit time 539 CPU ticks 3.208333248 uS
2024-12-01_21_48_12_865: EngineState: Total pulses 404691
2024-12-01_21_48_12_865: EngineState: Last valid fast msg Status 0x0 Sig0 0xAD2 Sig1 0x52D
2024-12-01_21_48_12_867: EngineState: Restarts 0
2024-12-01_21_48_12_867: EngineState: Interval errors 4363 short, 474 long
2024-12-01_21_48_12_867: EngineState: Total frames 39831 with CRC error 46 (0.115487936%)
2024-12-01_21_48_12_874: EngineState: Total slow channel messages 0 with crc6 errors 0 (NaN%)
2024-12-01_21_48_12_875: EngineState: Sync errors 4363
2024-12-01_21_48_12_876: EngineState: --------------------
2024-12-01_21_48_12_878: EngineState: confirmation_sentinfo:8
```

dron0gus, 2024-12-05:
https://github.com/rusefi/rusefi/pull/7126

```
2024-12-06_01_33_42_324: EngineState: ---- SENT input 1 ---- on 24A - Hall Input 3 / SENT
2024-12-06_01_33_42_325: EngineState: Unit time 353 CPU ticks 1.634259200 uS
2024-12-06_01_33_42_326: EngineState: Total pulses 72414
2024-12-06_01_33_42_327: EngineState: Last valid fast msg Status 0x0 Sig0 0xAD8 Sig1 0x527
2024-12-06_01_33_42_328: EngineState: Restarts 3
2024-12-06_01_33_42_329: EngineState: Interval errors 3721 short, 321 long
2024-12-06_01_33_42_331: EngineState: Total frames 3721 with CRC error 21 (0.564364416%)
2024-12-06_01_33_42_332: EngineState: Total slow channel messages 0 with crc6 errors 0 (NaN%)
2024-12-06_01_33_42_338: EngineState: Sync errors 3721
2024-12-06_01_33_42_339: EngineState: --------------------
2024-12-06_01_33_42_341: EngineState: confirmation_sentinfo:8
```

rusefillc, 2024-12-06:
[image]

commit 24ca7512 (2024-11-29): TIM1_CH4 ICU fix #7076
closed 2024-12-06 by rusefillc