rusefi/rusefi #7077: revisit SENT quality

rusefillc, 2024-11-15:
We are loosing some percentage of SENT packets WHY?

IRQ priority

rusefillc, 2024-11-19:
@dron0gus 

https://github.com/rusefi/rusefi/wiki/Developer-Performance-Tracing

[image]

[image]

[ignition_off_rusEFI_trace.zip](https://github.com/user-attachments/files/17820847/ignition_off_rusEFI_trace.zip)

rusefillc, 2024-12-06:
@dron0gus are we happy? close or else?

dron0gus, 2024-12-06:
@rusefillc not yet.
To make SENT robust we need to guaranty that we are able to handle timer interrupts as fast as 36uS (12 SENT ticks minimum pulse @ 3uS/pulse).
Now something is spending more in more priority IRQ or in a critical zone.

dron0gus, 2024-12-09:
Some random statistics on [lateDelay](https://github.com/rusefi/rusefi/blob/e3d3fd4218de6a20b91dfe6fb90e7c70b73c77be/firmware/controllers/system/timer/single_timer_executor.cpp#L43) vs HW overrun errors on SENT capture timer.
Default 8 uS
```
2024-12-09_01_03_27_781: EngineState: ---- SENT input 1 ---- on 24A - Hall Input 3 / SENT
2024-12-09_01_03_27_782: EngineState: Unit time 353 CPU ticks 1.634259200 uS
2024-12-09_01_03_27_783: EngineState: Pause pulse detected No
2024-12-09_01_03_27_784: EngineState: Total pulses 806328
2024-12-09_01_03_27_784: EngineState: Last valid fast msg Status 0x0 Sig0 0xAD6 Sig1 0x529
**2024-12-09_01_03_27_788: EngineState: HW overflows 7214** 
2024-12-09_01_03_27_792: EngineState: Pause pulses 5264 
2024-12-09_01_03_27_792: EngineState: Restarts 0
2024-12-09_01_03_27_794: EngineState: Interval errors 4923 short, 12157 long
2024-12-09_01_03_27_795: EngineState: Total frames 69077 with CRC error 1966 (2.846099392%)
2024-12-09_01_03_27_796: EngineState: Total slow channel messages 0 with crc6 errors 0 (NaN%)
2024-12-09_01_03_27_797: EngineState: Sync errors 4932
2024-12-09_01_03_27_797: EngineState: --------------------
2024-12-09_01_03_27_798: EngineState: confirmation_sentinfo:8
```
```
(gdb) p maxExecuteCounter
$28 = 19
```

4uS
```
2024-12-09_01_01_02_941: EngineState: ---- SENT input 1 ---- on 24A - Hall Input 3 / SENT
2024-12-09_01_01_02_942: EngineState: Unit time 354 CPU ticks 1.638888832 uS
2024-12-09_01_01_02_942: EngineState: Pause pulse detected No
2024-12-09_01_01_02_942: EngineState: Total pulses 830641
2024-12-09_01_01_02_943: EngineState: Last valid fast msg Status 0x0 Sig0 0xAD6 Sig1 0x529
**2024-12-09_01_01_02_943: EngineState: HW overflows 3750** 
2024-12-09_01_01_02_943: EngineState: Pause pulses 4426 
2024-12-09_01_01_02_945: EngineState: Restarts 0
2024-12-09_01_01_02_946: EngineState: Interval errors 3870 short, 11245 long
2024-12-09_01_01_02_950: EngineState: Total frames 72511 with CRC error 2079 (2.867151232%)
2024-12-09_01_01_02_952: EngineState: Total slow channel messages 0 with crc6 errors 0 (NaN%)
2024-12-09_01_01_02_954: EngineState: Sync errors 3871
2024-12-09_01_01_02_954: EngineState: --------------------
```
```
(gdb) p maxExecuteCounter
$27 = 20
```

2uS
```
2024-12-09_01_06_26_069: EngineState: ---- SENT input 1 ---- on 24A - Hall Input 3 / SENT
2024-12-09_01_06_26_070: EngineState: Unit time 354 CPU ticks 1.638888832 uS
2024-12-09_01_06_26_070: EngineState: Pause pulse detected No
2024-12-09_01_06_26_071: EngineState: Total pulses 814217
2024-12-09_01_06_26_072: EngineState: Last valid fast msg Status 0x0 Sig0 0xAD6 Sig1 0x529
**2024-12-09_01_06_26_073: EngineState: HW overflows 7333** 
2024-12-09_01_06_26_074: EngineState: Pause pulses 5221 
2024-12-09_01_06_26_075: EngineState: Restarts 0
2024-12-09_01_06_26_076: EngineState: Interval errors 4905 short, 13788 long
2024-12-09_01_06_26_079: EngineState: Total frames 67509 with CRC error 2357 (3.491386400%)
2024-12-09_01_06_26_079: EngineState: Total slow channel messages 0 with crc6 errors 0 (NaN%)
2024-12-09_01_06_26_084: EngineState: Sync errors 4911
2024-12-09_01_06_26_084: EngineState: --------------------
2024-12-09_01_06_26_085: EngineState: confirmation_sentinfo:8
```