rusefi/rusefi #7080: K6A Trigger

EA11R, 2024-11-18:
Update of the past issue , we finally have the scope information with spark plugs and without using logic 1 software from saeale

EA11R, 2024-11-18:
Refer to https://github.com/rusefi/rusefi/pull/6796

huntertrainboy, 2024-12-11:
@rusefillc any update on this? I apologize it took so long to get the logic files(it's my car). Is there enough info from the files captured to setup a proper trigger? Thank you. [logic analyzer files](https://github.com/user-attachments/files/17799489/ea21rlogic.zip)

rusefillc, 2024-12-18:
@FDSoftware can you help?

FDSoftware, 2024-12-18:
@huntertrainboy @EA11R 
[image]
I'm going to need you to check the wiring of both the logic analyzer and the VR sensor, I see some random pulses that don't correspond to the shape of the trigger. After doing so, try to get the captures from the logic analyzer back, the same method that Andrey requested, with and without the spark plugs connected.

huntertrainboy, 2024-12-18:
@FDSoftware Thanks for looking at it. I will check the wiring again. The car runs on its stock computer with no issues. I have to reverse the polarity from the connector, but I will check that it was fully inserted and get another 2 files sent over Friday evening.

EA11R, 2024-12-21:
@FDSoftware @huntertrainboy  has the engine and car however I have the distributor wheel only,  I can spin that with a drill and get a clear logic file but that wouldn't be enough to simulate ignition events 

I have the idea of connecting multiple logic pins to the injector , ignition pins of the orignal ecu and the reluctor signal after oem conditioner to get the timing of events 

Is it possible to do it this way as I haven't done it this way before