rusefi/rusefi #7085: LUA script size severely limited

ggurov, 2024-11-19:
**Describe the bug**
LUA is out of memory with an unreasonably short script

**Hardware**
 - Board: mega100
 - Engine info N/A

**To Reproduce**
Steps to reproduce the behavior:
1. Go to console
2. paste code into lua
3. click write to ecu
4. See error in console

```
2024-11-19_17_53_42_122: EngineState: confirmation_luareset:8
2024-11-19_17_53_42_218: EngineState: LUA: Tearing down instance...
2024-11-19_17_53_42_218: EngineState: LUA loading script length: 401...
2024-11-19_17_53_42_218: EngineState: LUA registered sensor of type Rpm
2024-11-19_17_53_42_218: EngineState: Added Lua CAN RX filter id 0x201 mask 0x1FFFFFFF without custom function
2024-11-19_17_53_42_219: EngineState: LUA script loaded successfully!
2024-11-19_17_53_42_219: EngineState: Lua memory heap usage: 14414 / 18000 bytes = 80.0%
2024-11-19_17_53_45_842: EngineState: TS -> Page 0 write chunk offset 5696 count 1024 (output_count=7244)
2024-11-19_17_53_45_842: EngineState: TS -> Page 0 write chunk offset 6720 count 1024 (output_count=7244)
2024-11-19_17_53_45_842: EngineState: TS -> Page 0 write chunk offset 7744 count 1024 (output_count=7244)
2024-11-19_17_53_45_842: EngineState: TS -> Page 0 write chunk offset 8768 count 1024 (output_count=7244)
2024-11-19_17_53_45_842: EngineState: TS -> Page 0 write chunk offset 9792 count 1024 (output_count=7244)
2024-11-19_17_53_45_842: EngineState: TS -> Page 0 write chunk offset 10816 count 1024 (output_count=7244)
2024-11-19_17_53_45_842: EngineState: TS -> Page 0 write chunk offset 11840 count 1024 (output_count=7244)
2024-11-19_17_53_45_843: EngineState: TS -> Page 0 write chunk offset 12864 count 832 (output_count=7244)
2024-11-19_17_53_45_843: EngineState: confirmation_luareset:8
2024-11-19_17_53_45_844: EngineState: LUA: Tearing down instance...
2024-11-19_17_53_45_845: EngineState: LUA loading script length: 402...
2024-11-19_17_53_45_845: EngineState: LUA ERROR loading script: not enough memory
2024-11-19_17_53_45_850: EngineState: LUA: Tearing down instance...
2024-11-19_17_53_45_850: EngineState: LUA MEMORY LEAK DETECTED: 160 bytes used after teardown
```

**Expected behavior**
Being able to write decent sized scripts in LUA

rusefillc, 2024-11-19:
```
function onTick()
end
```

Lua memory heap usage: 12988 / 18000 bytes = 72.1%

rusefillc, 2024-11-20:
Same script with https://github.com/rusefi/rusefi/commit/0d333248e3b5c82735d5fdeea67c13fa49ea66a7
``2024-11-19_18_59_24_653: EngineState: Lua memory heap usage: 12869 / 18000 bytes = 71.4%``

116 bytes for three method declarations, let's assume 40 bytes per method declaration.

ggurov, 2024-11-20:
```
n = 0
setTickRate(5) -- set tick rate to 5hz

rpm = Sensor.new("RPM")
rpm:setTimeout(3000)
-- clt = Sensor.new("CLT")
-- tps = Sensor.new("TPS1")
-- pps = Sensor.new("AcceleratorPedal")
map = Sensor.new("map")
map:setTimeout(3000)

-- canRxAdd(0x200)
canRxAdd(0x201)
canRxAdd(0x203)

function onCanRx(bus , id, dlc, data)
    -- print('got CAN id=' ..id ..' dlc=' ..dlc)

    if id == 0x201 then
        rpm : set((data[2] * 256) + data[1])
    end

      if id == 0x203 then 
            map : set( (data[1] + (data[2]*256)) * (2000.0 / 65536.0 ))
    end 
    
end
```
--->

```
2024-11-19_19_56_23_198: EngineState: confirmation_luareset:8
2024-11-19_19_56_23_198: EngineState: LUA loading script length: 554...
2024-11-19_19_56_23_198: EngineState: LUA registered sensor of type Rpm
2024-11-19_19_56_23_198: EngineState: LUA registered sensor of type Map
2024-11-19_19_56_23_200: EngineState: Added Lua CAN RX filter id 0x201 mask 0x1FFFFFFF without custom function
2024-11-19_19_56_23_200: EngineState: Added Lua CAN RX filter id 0x203 mask 0x1FFFFFFF without custom function
2024-11-19_19_56_23_200: EngineState: LUA script loaded successfully!
2024-11-19_19_56_23_200: EngineState: Lua memory heap usage: 14700 / 18000 bytes = 81.6%
```

removing 2 lines from the top:
```
rpm = Sensor.new("RPM")
rpm:setTimeout(3000)
-- clt = Sensor.new("CLT")
-- tps = Sensor.new("TPS1")
-- pps = Sensor.new("AcceleratorPedal")
map = Sensor.new("map")
map:setTimeout(3000)

-- canRxAdd(0x200)
canRxAdd(0x201)
canRxAdd(0x203)

function onCanRx(bus , id, dlc, data)
	-- print('got CAN id=' ..id ..' dlc=' ..dlc)

	if id == 0x201 then
		rpm : set((data[2] * 256) + data[1])
	end

      if id == 0x203 then 
            map : set( (data[1] + (data[2]*256)) * (2000.0 / 65536.0 ))
    end 
	
end
```

results in 
```2024-11-19_19_58_14_557: EngineState: confirmation_luareset:8
2024-11-19_19_58_14_656: EngineState: LUA: Tearing down instance...
2024-11-19_19_58_14_656: EngineState: LUA MEMORY LEAK DETECTED: 128640 bytes used after teardown
2024-11-19_19_58_14_656: EngineState: LUA loading script length: 509...
2024-11-19_19_58_14_656: EngineState: LUA ERROR loading script: not enough memory
2024-11-19_19_58_14_656: EngineState: LUA: Tearing down instance...
2024-11-19_19_58_14_656: EngineState: LUA MEMORY LEAK DETECTED: 200 bytes used after teardown
```

rusefillc, 2024-11-20:
@ggurov please pull latest / update your FW, I've removed some functions

See https://github.com/rusefi/fw-custom-mega100/commit/fd924d618332f9d121c8641575947d636624797a

```
2024-11-19_19_59_22_891: EngineState: LUA script loaded successfully!
2024-11-19_19_59_22_891: EngineState: Lua memory heap usage: 10583 / 18000 bytes = 58.7%
```

overall just use 427 chip. there is only that much we can gain by reducing less needed functionality

rusefillc, 2024-11-20:
Some fat was trimmed from uaefi and mega100

closed 2024-11-20 by rusefillc