rusefi/rusefi #7093: Critical error when using big values on Boost Control Open Loop with Y axis as MAP

ElDominio, 2024-11-22:
**Describe the bug**
Error in the picture

**To Reproduce**
Steps to reproduce the behavior:
1. Use values over big values on Y axis of boost controll open loop

**Expected behavior**
I can use correct values

**Screenshots**

[image]

ElDominio, 2024-11-25:
Settings Y axis to MAP does not work, and using values like these produces this error I can't understand. Why is it expecting my boost load to be under 4?

rusefillc, 2024-11-25:
255 represented as 255

256 represented as 0, looks like int8 value axis

[image]

commit 2de5e73d (2024-11-25): Critical error when using big values on Boost Control Open Loop with Y axis as MAP fix #7093