rusefi/rusefi #7109: How does ETB-style wastegate actuator work?

ElDominio, 2024-11-29:
**Describe the bug**
I have activated all related I/O, do i just fill out the boost target table and hope for it to work? Are there any gauges? any range tests?

**Hardware**
 - Board: [h154]
 - Engine info [Theta2]

**Expected behavior**
An option to explicitly test if it works, explicitly state feed-forward duty, gauges to check current duty. The same useability as throttle

**Screenshots**
[image]

rusefillc, 2024-12-04:
@ElDominio great top level ask!

@ElDominio need your help, can you please call 1-800-RE-business-analyst and work with them on specific actionable items here?

ElDominio, 2024-12-04:
- feed forward open loop wastegate position table
- enable etb style wastegate under boost control
- automatic etb range calibration (auto calibrate etb button for wastegate)
- auto pid calibration for wastegate

ElDominio, 2024-12-09:
[image]

seems like i will be asked about this weekly now

rusefillc, 2024-12-09:
@ElDominio this is a wide ask here, in current scope it might be NOT done soon

if you have urgency you either have to inject resources or reduce ask by focusing on the 20% which gives 80% of the result

ElDominio, 2024-12-09:
Defnitiely do not need automatic anything, definitely need base table for position and etb style wastegate allowed as a boost control method

rusefillc, 2024-12-09:
https://github.com/rusefi/rusefi/issues/7150

ElDominio, 2025-05-15:
Current implementation doesn't let the user select if we are using E-wastegate; we can control it since Auto-Calibrate works, but we are not telling the boost control system to use ewasategate, so moving the open loop position does nothing

[image]