rusefi/rusefi #7117: Clutch switch should be able to be shared by Launch Control and Torque Reduction

ElDominio, 2024-12-02:
**Describe the bug**
This error should not appear when these three functions share a pin (Clutch Switch Input, Launch Control, and Torque Reduction)

**Hardware**
 - Board: h154
 - Engine info bk theta

**To Reproduce**
Steps to reproduce the behavior:
Set clutch switch as both clutch switch input inside chassis sensors, and as torque reduction input

**Expected behavior**
No error, since the same sensor is used for different functions

**Screenshots**
[image]

rusefillc, 2024-12-02:
1) @ElDominio you are in violation of https://en.wikipedia.org/wiki/XY_problem
2) @ElDominio please RTFM

[image]

ElDominio, 2024-12-12:
This works for Launch, does NOT work for shift torque since Clutch Down Switch is not an option @rusefillc

ElDominio, 2024-12-12:
[image]
 and this just happened

ElDominio, 2024-12-12:
XY:

I am trying to share Clutch Input Switch defined in Chassis Sensors to be used for Shift Torque Reduction and Launch Control

closed 2024-12-16 by kifir23917

kifir23917, 2024-12-16:
@ElDominio, I have added `Cluch Down Switch` as an `Activation Mode` option on `Shift Torque Reduction (Flat Shift)` dialog

[image]

ElDominio, 2024-12-16:
Both clutch up and down should be options. Some people use up some people use down

kifir23917, 2024-12-18:
I have added `Clutch Up Switch` option:

[image]

kifir23917, 2024-12-18:
BTW: `Launch Control` doesn't provide `Clutch Up Switch` option for its `Activation Mode`:

[image]

Is it OK?

ElDominio, 2024-12-18:
> BTW: `Launch Control` doesn't provide `Clutch Up Switch` option for its `Activation Mode`:
> 
> ![image](https://private-user-images.githubusercontent.com/159958110/396807646-c3e3f2d4-24f7-450b-9892-70feba72e713.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ1MDgzNTUsIm5iZiI6MTczNDUwODA1NSwicGF0aCI6Ii8xNTk5NTgxMTAvMzk2ODA3NjQ2LWMzZTNmMmQ0LTI0ZjctNDUwYi05ODkyLTcwZmViYTcyZTcxMy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjE4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIxOFQwNzQ3MzVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zNzFkYTIzYTkyM2VjMTkyMzc0YjY5NmQ4ZDFjMTk0NmZiOTBmNWQ1NzM2MGI4ZTVjNDNhMmRlNTVhOTczYmJmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.zIIprgsKAZO17wJpSXnIlKhbxHJOT9i3uuz8BcfQQ8I)
> 
> @ElDominio , is it OK?

Should have "Clutch Up Switch"  also selectable

> I have added `Clutch Up Switch` option: [...]

This one is perfect 

"Mode" should be hidden on both Launch and Torque reduction

kifir23917, 2024-12-18:
> "Mode" should be hidden on both Launch and Torque reduction
@ElDominio, is my understanding correct that we should get rid of `Launch Button Mode` setting on `Launch Control` dialog as well?

[image]

rusefillc, 2024-12-19:
Related to https://github.com/rusefi/rusefi/issues/6951

commit 53cc9049 (2024-12-16): Add `Clutch Down Switch` option to`Activation Mode` combobox on `Shift Torque Reduction (Flat Shift)` dialog (closes #7117)