rusefi/rusefi #7118: Ethernet broken in latest release

joesphan, 2024-12-04:
**Describe the bug**
It seems that ethernet might be broken in latest release.

**Hardware**
 - Board: STM32F767ZI nucleo bare board

**To Reproduce**
Steps to reproduce the behavior:
1. Download rusEFI master.2024.11.22.stm32f767_nucleo.4047957739 onto f767 with ethernet
2. plugin ethernet directly to pc and set 192.168.10.10 as ip, 192.168.10.1 as gateway, default netmask 
3. No ping, but TunerStudio connects with 192.168.10.1 port 29000
4. so that is the working version. 
5. if you download latest,  rusEFI master.2024.12.02.stm32f767_nucleo.4278423622
6. repeat the steps, will result in no ping no tunerstudio connection

**Expected behavior**
Ethernet to connect

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]
[image]
[image]

**Additional context**

joesphan, 2024-12-04:
Upon further inspection it looks like the latest release is broken in general. no usb connection either.

rusefillc, 2024-12-04:
https://github.com/rusefi/rusefi/issues/6242 is a recent change specifically for that board

rusefillc, 2024-12-04:
@FDSoftware is your unit affected by this?

FDSoftware, 2024-12-04:
> @FDSoftware is your unit affected by this?

yep, I have the same issue, later I will try adding a commit reverting 45ac95e

joesphan, 2024-12-05:
Just spitballing here, it seems like larger flash size might eat into bootloader space, can this be the case?

rusefillc, 2024-12-05:
revert of 2mb did not help me
I've reverted openblt and things are blinking blue happily

i will now re-apply 2mb without openblt

rusefillc, 2024-12-05:
2mb re-enabled, things seem to blink as expected

Separate https://github.com/rusefi/rusefi/issues/7124 follow-up ticket for later

closed 2024-12-05 by rusefillc

joesphan, 2024-12-05:
X2 Happy to report it works now too