rusefi/rusefi #7123: Target AFR blend feature

rusefillc, 2024-12-05:
Users would like to add/remove fuel at a certain temperature such when the bike is too hot in order to quench the motor. With closed loop control that means Target AFR blend feature

rusefillc, 2024-12-05:
@ElDominio what did you say about having four of these? Do you mean four different tables applied consecutively one by one?

I guess let's start with one configurable for now.

benassi124, 2024-12-05:
if i recall correctly he wanted essentially a blend table but instead of modifying VE it would be modifying lambda target

benassi124, 2024-12-08:
@ElDominio It would be nice to also have a target lambda blend table for Ethanol %

ElDominio, 2024-12-08:
if there is a generic blend table, you can choose your blend parameter as Ethanol%, making it exactly what you ask for

rusefillc, 2024-12-15:
@ElDominio @kifir23917 https://github.com/rusefi/rusefi/issues/7170

rusefillc, 2024-12-15:
https://github.com/rusefi/rusefi/pull/7171

rusefillc, 2024-12-17:
@kifir23917 do we have any coverage for any blend at all? if not please create a separate ticket

closed 2024-12-17 by rusefillc