rusefi/rusefi #7131: 8chan yaml is a mess

rusefillc, 2024-12-06:
Feedback from email

3. PINOUT - the pinouts include suggested functions of each pin.  This is a big no-no in my opinion, and most of your competition doesn't do this (the less professional the system, the more likely they are to pre-label pins).  It might be holdover from the plug and play boards where you have to match existing functionality, but with a universal board, users expect them to be universal.  I already know what my motor needs/expects, I just want a wiring diagram that shows me where the analogs, digitals, and outputs are. So to that end...
ACTION ITEMS:
a. All "Aux Analog #" inputs should be renamed "Analog #".  Set type to "av"
b. Pins 2A, 31A, 32A, 33A, 34A should be renamed "Analog #". Set type to "av"
c. Pins 27A and 29A should be renamed "Ana Temp 3" and "Ana Temp 4". Set type to "at"
d. Pins 28A and 30A should be renamed "Ana Temp 1" and "Ana Temp 2" (replacing Aux with Ana).  Set type to "at"
e. Pins 21A and 26A should be renamed a Hall Input (I first read VSS thinking it was ground pin for hall sensors....).
f. Pin 24A - it looks like SENT is configurable in TunerStudio, so does this pin really need to be labeled the SENT pin?  If not, remove notation.
g. Pins 9A, 13A, 15A, 23A, 24A, 26A should have type set to "hall"
h. Pin 1B should have type set to "hall/vr"
i. Pins 5A, 6A, 14A, and 22A should have type set to "switch".
j. Rename 2C/4C to DC4+/DC4-
k. Rename 8D, 9D, 10D, 11D, 12D, 14D, 15D, 20D, 21D, 22D to "Low Side #".  Set type to "ls" or "aux_ls"? Along with the 7 preexisting low side drivers.
l. Update Wiki to mention 17 low side drivers and 3 high side drivers (though OMG I would have cut a few of those and traded them for more analog or hall inputs).
m. Update Wiki to mention 4 temperature inputs (not 2).
n. Update Wiki to mention 7 Hall inputs (VSS and Flex are Hall inputs).

4. PINOUT - pins should be reasonably consistent on what pull up/down they use (on a universal board).  It should also be easy to find out what they are.  Instead I have to cross reference the schematic and the BOM.
ACTION ITEMS:
a. On the wiki, or maybe under "Typical Function", denote pullup/pulldown specifications for pins/pin types.
b. Pins 1A-4A, 10A-12A, 19A-20A, 31A, 32A, 33A, 34A should be labeled "680k pulldown"
c. Pins 28A and 30A should be labeled "4.7k pullup"
d. Pins 27A and 29A should be labeled "optional group 2.2k pullup".
e. Pins 9A, 13A, 15A, 21A, 23A, 24A, 26A have either 4.5k up or 3.3k down, user selectable.
f. Pins 5A, 6A, 14A, 22A have... ?? 4.7k pullup + LED which makes it not really 4.7.  (Why do you have LEDs on the board?)
g. Pins 5C, 6C, 7C have similar 4.7k+LED pulldown.
h. All 17 low sides (don't make me repeat them!) have 4.7k+LED pull up to 12V, EXCEPT for pin 15D, "tach" output, which seems to have a pullup to 5v?  Not sure why it has two diodes in the same direction? (p15 of the schematic, OUT_TACH signal).

rusefillc, 2025-03-11:
@ElDominio do you know anything about this? close or break apart into actionable tickets or else?

ElDominio, 2025-03-11:
While I can see the emailer's point, right now sales aren't so high that this needs to be focus. My 8chans on hand are all being used to develop new integrations and as such, giving this priority does not make sense. I think this can be actioned upon when it's a more pressing issue (right now there are no installers buying this and being confused about it)