rusefi/rusefi #7133: wheel speed input

rusefillc, 2024-12-06:
Is one channel hall wheel speed input even possible as is, with X tooth per wheel revolution in terms of IRQ rate?

What about four channels of same? Do we _have_ to use https://github.com/rusefi/rusefi-hardware/tree/main/frequency-divider or would we have to use external hardware with digital communication?

racer-coder, 2024-12-06:
Having some sort of limit is ok, esp if published.  Link G4+ (older 16-bit platform) published a limit of 500Hz for digital inputs 1-8 and 6500Hz for inputs 11-16 (only available on G4+ Thunder if I remember).  The stated policy was "you can exceed it as your own risk but you may exceed processing capabilities of the ECU".  Emtron says 30kHz on all inputs!!!  Yikes.  MaxxECU says 10kHz limit.

48 teeth per rotation * 834 rotations per mile * 150 miles/hour / 3600 seconds/hour = 1668 Hz.
I believe turbo sensors are a bigger issue at 150k rpm = 2500Hz at one tooth per revolution.  But they usually measure compressor blades (or turbine blades), so, uh, multiply by anywhere from 7-11.

If you use the DMA to a FIFO trick and batch process the results (maybe software tick reducer) then you should be able to handle them no problem, you wouldn't need an IRQ to process them realtime, just batch process them as part of your main tick loop (thus avoiding separate IRQ overhead etc.)  More specifically: EXTI signals DMA to execute, DMA reads from the port, writes to a fifo, reads from the current time value, writes to a fifo.  That way you have timestamped event log when you go to read it later.  I don't remember how this would fit into your codebase, I think at one point you were using EXTI but then moved away from it, and I don't know that you were ever using DMA.

FDSoftware, 2024-12-19:
> Is one channel hall wheel speed input even possible as is, with X tooth per wheel revolution in terms of IRQ rate?

if signal shape is not important, we can use one timer on PWM Input mode:
[image]

supported timers are:
TIM1 to TIM5, TIM8, TIM9, TIM12 

on these targets:
STM32F40x, 32F7x and H74x|H75x

This way the only IRQ will be the period for which you want to read the DMA variable (I think in our case we could read the CCR1 without even using a DMA channel [HAL has `HAL_TIM_ReadCapturedValue`, i need to find the alike in chibios])

some info about this:
https://blog.embeddedexpert.io/?p=2019
https://stackoverflow.com/questions/41262460/stm32f4-pwm-input-capture-of-high-frequency-signal-approx-2mhz

> What about four channels of same? Do we _have_ to use https://github.com/rusefi/rusefi-hardware/tree/main/frequency-divider or would we have to use external hardware with digital communication?

Thinking very randomly, I would vote for an external HW, perhaps some low pin count STM32F4X, (could it also handle the ckp/cmp inputs?, handle the injectors/coils?)