rusefi/rusefi #7134: Stepper idle useability

ElDominio, 2024-12-07:
**Is your feature request related to a problem? Please describe.**
 I'm always frustrated when I don't know how many steps are being commanded.  Additional to idle output %, can we get # of steps? Where 0 is 0 steps, and 100% is the max number of steps entered in Idle Settings 

**Would we need new configuration fields?**
New "# of steps" gauge

**Describe the solution you'd like**
Add a # of steps" gauge

**Describe alternatives you've considered**
Ugly, difficult math

ElDominio, 2024-12-15:
@rusefillc  easier ticket name, easier to fulfil?

rusefillc, 2024-12-15:
> @rusefillc easier ticket name, easier to fulfil?

@ElDominio please confirm that I should drop the 60-2 and focus on stepper gauge?

rusefillc, 2024-12-18:
@FDSoftware any chance you could?

FDSoftware, 2024-12-18:
[image]
something like this would be ok? @ElDominio

ElDominio, 2024-12-18:
Yep perfect

FDSoftware, 2024-12-18:
@ElDominio when today's Nightly is generated you will be able to test this (or use the bin from: [d2a0fd1](https://github.com/rusefi/rusefi/commit/d2a0fd18941c48fff5b8b0544dd08100f05b19e5))

rusefillc, 2024-12-18:
> @ElDominio when today's Nightly is generated you will be able to test this (or use the bin from: [d2a0fd1](https://github.com/rusefi/rusefi/commit/d2a0fd18941c48fff5b8b0544dd08100f05b19e5))

I think the is no nightly on main repo, only snapshot builds right after the fact

FDSoftware, 2024-12-18:
@rusefillc 
`.github/workflows/build-firmware.yaml` defines a cronjob that runs every day from this build to today's date:

https://github.com/rusefi/rusefi/releases/tag/2024-03-27
https://github.com/rusefi/rusefi/releases/tag/2024-12-18

```
on:
  push:
  pull_request:
  schedule:
    # this produces GitHub nightly releases https://github.com/rusefi/rusefi/releases
    # by the way GitHub actions use UTC https://www.utctime.net/ which matches GMT time
    # See https://github.com/rusefi/rusefi/issues/6286 for notes on scopes enabled on NIGHTLY_RELEASE_ACCESS_TOKEN
    - cron: '27 0 * * *'
```

I see that #6286 warns that they were not being generated, but since that issue we have 28 release pages on https://github.com/rusefi/rusefi/releases

rusefillc, 2024-12-18:
@FDSoftware yes, you are right and I stay corrected - there is a nightly process

but I bet you that's not the way @ElDominio consumes binary: @ElDominio probably grabs snapshots from build server https://rusefi.com/build_server/ either manfully or with console auto update mechanism which downloads from https://rusefi.com/build_server/

ElDominio, 2024-12-18:
Can confirm what @rusefillc said

FDSoftware, 2025-01-03:
@rusefillc this one is also done

closed 2025-01-03 by rusefillc