rusefi/rusefi #7140: Burning lua breaks CLT multifplier CLT axis

ElDominio, 2024-12-08:
**Describe the bug**
I have big custom lua, and when I burn it, my CLT temp axis gets corrupted.

**Hardware**
 - Board: Platinum revC F4
 - Engine info Ford Voodoo

**To Reproduce**
Steps to reproduce the behavior:
1. Write very large Lua (: 24381 / 60000 bytes = 40.6%)
2. Hit burn ECU
3. FATAL shows up because of corrupted CLT axis

**Expected behavior**
Not this error

**Screenshots**
[image]

ElDominio, 2024-12-09:
https://gist.github.com/ElDominio/5439432fca80bac930db16fb02756be2

problem lua script

rusefillc, 2024-12-09:
@dron0gus are you able to reproduce? sounds like out of range memory access?

dron0gus, 2024-12-12:
@rusefillc 
```
2024-12-12_15_07_17_692: EngineState: LUA loading script length: 3095...
2024-12-12_15_07_17_693: EngineState: LUA ERROR loading script: not enough memory
2024-12-12_15_07_17_693: EngineState: LUA: Tearing down instance...
```
On my F4 (F427) Proteus with provided LUA script.

@ElDominio what firmware version are you using? Is it some special build?

ElDominio, 2024-12-12:
[image]

dron0gus, 2024-12-13:
@ElDominio can you share your tune?

dron0gus, 2024-12-13:
```
Starting Lua interpreter...
Lua memory heap usage: 16 / 25000 bytes = 0.0%
Done starting Lua interpreter...
Lua memory heap usage: 3480 / 25000 bytes = 13.9%
Loading LUA libs...
Done loading LUA libs...
Lua memory heap usage: 5288 / 25000 bytes = 21.1%
Loading RusEFI hooks...
Done loading RusEFI hooks...
Lua memory heap usage: 15216 / 25000 bytes = 60.8%
```

ElDominio, 2024-12-13:
https://rusefi.com/online/view.php?msq=1721

dron0gus, 2024-12-16:
Even I was not able to reproduce exactly the same error I get:
`EngineState: CRITICAL error: a NaN getWrappedAdvance 20241127@proteus`
With your Lua script.
Funny thing it was not executed due to "not enough memory" and failed to fully release allocated:
```
2024-12-16_21_59_45_985: EngineState: LUA loading script length: 3094...
2024-12-16_21_59_45_986: EngineState: LUA ERROR loading script: not enough memory
2024-12-16_21_59_45_987: EngineState: LUA: Tearing down instance...
2024-12-16_21_59_45_988: EngineState: LUA MEMORY LEAK DETECTED: 4096 bytes used after teardown
```
Something is corrupting engine during Lua deinitialization:
```
(gdb) p ___engine.ignitionState
$16 = {<ignition_state_s> = {baseDwell = 0, sparkDwell = 0, dwellDurationAngle = 0, cltTimingCorrection = {<scaled_channel_base> = {<No data fields>}, {m_value = 0, m_firstByte = 0 '\000'}}, 
    timingIatCorrection = {<scaled_channel_base> = {<No data fields>}, {m_value = 0, m_firstByte = 0 '\000'}}, timingPidCorrection = {<scaled_channel_base> = {<No data fields>}, {m_value = 0, 
        m_firstByte = 0 '\000'}}, baseIgnitionAdvance = {<scaled_channel_base> = {<No data fields>}, {m_value = 0, m_firstByte = 0 '\000'}}, 
    correctedIgnitionAdvance = {<scaled_channel_base> = {<No data fields>}, {m_value = 0, m_firstByte = 0 '\000'}}, tractionAdvanceDrop = {<scaled_channel_base> = {<No data fields>}, {m_value = 0, 
        m_firstByte = 0 '\000'}}, dwellVoltageCorrection = 0, luaTimingAdd = 0, luaTimingMult = 0, luaIgnitionSkip = false, unusedBit_13_1 = false, unusedBit_13_2 = false, unusedBit_13_3 = false, 
    unusedBit_13_4 = false, unusedBit_13_5 = false, unusedBit_13_6 = false, unusedBit_13_7 = false, unusedBit_13_8 = false, unusedBit_13_9 = false, unusedBit_13_10 = false, unusedBit_13_11 = false, 
    unusedBit_13_12 = false, unusedBit_13_13 = false, unusedBit_13_14 = false, unusedBit_13_15 = false, unusedBit_13_16 = false, unusedBit_13_17 = false, unusedBit_13_18 = false, unusedBit_13_19 = false, 
    unusedBit_13_20 = false, unusedBit_13_21 = false, unusedBit_13_22 = false, unusedBit_13_23 = false, unusedBit_13_24 = false, unusedBit_13_25 = false, unusedBit_13_26 = false, unusedBit_13_27 = false, 
    unusedBit_13_28 = false, unusedBit_13_29 = false, unusedBit_13_30 = false, unusedBit_13_31 = false}, <No data fields>}
```
IdleController somehow setting -nan to luaTimingAdd and corrupts some more surrounding. But this one looks like ricochet from some previous memory corruption.
```
0x0801fcca in Engine::resetLua (this=0x10000000 <___engine>) at ./controllers/algo/engine.cpp:321
321		module<IdleController>().unmock().luaAdd = 0;
(gdb) p ___engine.ignitionState
$17 = {<ignition_state_s> = {baseDwell = 4, sparkDwell = 4, dwellDurationAngle = 0, cltTimingCorrection = {<scaled_channel_base> = {<No data fields>}, {m_value = 0, m_firstByte = 0 '\000'}}, 
    timingIatCorrection = {<scaled_channel_base> = {<No data fields>}, {m_value = 0, m_firstByte = 0 '\000'}}, timingPidCorrection = {<scaled_channel_base> = {<No data fields>}, {m_value = 0, 
        m_firstByte = 0 '\000'}}, baseIgnitionAdvance = {<scaled_channel_base> = {<No data fields>}, {m_value = 608, m_firstByte = 96 '`'}}, 
    correctedIgnitionAdvance = {<scaled_channel_base> = {<No data fields>}, {m_value = 608, m_firstByte = 96 '`'}}, tractionAdvanceDrop = {<scaled_channel_base> = {<No data fields>}, {m_value = 0, 
        m_firstByte = 0 '\000'}}, dwellVoltageCorrection = 1, luaTimingAdd = -nan(0x7fffff), luaTimingMult = 1, luaIgnitionSkip = false, unusedBit_13_1 = false, unusedBit_13_2 = false, unusedBit_13_3 = false, 
    unusedBit_13_4 = false, unusedBit_13_5 = false, unusedBit_13_6 = false, unusedBit_13_7 = false, unusedBit_13_8 = false, unusedBit_13_9 = false, unusedBit_13_10 = false, unusedBit_13_11 = false, 
    unusedBit_13_12 = false, unusedBit_13_13 = false, unusedBit_13_14 = false, unusedBit_13_15 = false, unusedBit_13_16 = false, unusedBit_13_17 = false, unusedBit_13_18 = false, unusedBit_13_19 = false, 
    unusedBit_13_20 = false, unusedBit_13_21 = false, unusedBit_13_22 = false, unusedBit_13_23 = false, unusedBit_13_24 = false, unusedBit_13_25 = false, unusedBit_13_26 = false, unusedBit_13_27 = false, 
    unusedBit_13_28 = false, unusedBit_13_29 = false, unusedBit_13_30 = false, unusedBit_13_31 = false}, <No data fields>}
```

dron0gus, 2024-12-16:
Memory leak looks like bug in used memory calculations https://github.com/rusefi/rusefi/pull/7177

dron0gus, 2025-02-10:
Even CLT multiplier CLT axis is located right after lua script in settings I was to able to reproduce corruption of CLT axis values with Lua script manipulation.
Corruption causing NaN error (reported above) was caused by outdated compiler producing buggous code.

closed 2025-02-10 by rusefillc