rusefi/rusefi #7149: adjust default TS dashboard layout

rusefillc, 2024-12-09:
Can we embed custom background into .ini so that by default new projects have it? what about custom layout in terms of gauges/indicators?

rusefillc, 2024-12-09:
@ggurov do you know anything about it?

rusefillc, 2024-12-10:
```
[FrontPage]
	; Gauges are numbered left to right, top to bottom.
	;
	;	1  2  3  4
	;	5  6  7  8

	gauge1 = RPMGauge
	gauge2 = CLTGauge
	gauge3 = TPSGauge
	gauge4 = MAPGauge
#if LAMBDA
	gauge5 = lambda1Gauge
#else
	gauge5 = afr1Gauge
#endif
	gauge6 = VBattGauge
	gauge7 = dwellGauge
	gauge8 = correctedIgnAdvGauge

```

we have that but what about cooler layouts which are not boring 2x4 grid?

ggurov, 2024-12-10:
none of those are possible with free version of TS. 

there are some dashboards that are shipped with TS, that have signature of those controllers, ms1/ms2/ms3. And we could also ship a .dash file for those that have registered TS.

LT401vette, 2024-12-11:
The MS3-Pro products do this. 
They created a dash that is bundled with TS and when you create a new project if the signature represents a MS3-Pro product, that dash is used as the default dash. If not registered, you still see the normal 8 dial.