rusefi/rusefi #7150: DC motor wastegate position table feed forward open loop table

rusefillc, 2024-12-09:
6x6 1 byte S08 elements

ElDominio, 2024-12-20:
any progress on htis?

kifir23917, 2024-12-21:
@rusefillc, do we need increase this ticket priority?

ElDominio, 2025-01-04:
a comment was deleted is there anything on this?

ElDominio, 2025-03-12:
Tune at: https://rusefi.com/online/index.php?vehicleName=NegroWastegate&user_id=675&sorts[uploadedMsq]=-1&sorts[uploadedLog]=-1

Lua is inside tune, but important bit is:

wgPid = Pid.new(.01, 0.1, 0.005, -70, 70)
wgPid : setOffset(0.1)
wgPid : get(10, 10)
wgPid : reset()

startPwm(1, 500, 0.1) --pwm1 is pwm
startPwm(2, 100, 0) --pwm2 is dir
startPwm(3, 1, 0) --pwm3 is dis

wgPos = getSensor("WastegatePosition")
		targetWgPos = table3d(1, RPMread / 4, mapPres)
		-- p, i, d, min, max

		wgPidVal = wgPid : get(targetWgPos, wgPos)
		
		if wgPidVal < 0 then
			setPwmDuty(2, 0)
			wgPidVal = -wgPidVal
		else
			setPwmDuty(2, 1)
		end
		
		setPwmDuty(1, wgPidVal/10)
		wgPid : reset()

FDSoftware, 2025-04-05:
> 4x4 1 byte S08 elements
> feed forward open loop wastegate position table

@ElDominio I think the interface in TS needs to be made a little more intuitive; but currently, as soon as ETB-Wastegate is selected, the same table is used as feedforward as the PWM output, so there's no need to create a new one just for ETB.

Another additional thing is that we can now select both a PWM and ETB output at the same time to control the same Turbo, and I don't think that's a good idea!
See the images below:
[image]

[image]

I propose changing the `enable` option from a bool to an enum so that you can select:
* disabled
* enabled (PWM)
* enabled (E-Waste Gate)

Then with this, disable the PWM output both in TS and in the firmware where appropriate, and add a helper text on the boost control panel so that the user knows that they have to configure the e-waste gate h-bridge from the e-waste gate actuator tab.

rusefillc, 2025-04-05:
@FDSoftware yes, great idea! let's reuse existing PWM open loop table for DC motor feedforward!

@FDSoftware note that I've created smaller #7150 for the purpose of focusing on just the feedforward, separating that urgent effort from less urgent wider #7109

rusefillc, 2025-04-10:
first pair

[image]

FDSoftware, 2025-04-10:
> that's similar to how for ETB idle we have two feedforwards and two PIDs - one for desired IAC position, and another for PWM to hold that desired powition

Could it be that in both cases the second PID is the closed loop one?

[image]

rusefillc, 2025-04-10:
@ElDominio @Abricosvw we now have https://github.com/rusefi/rusefi/issues/7150!

open question if just a curve would do, or do we need a proper 3D table?

rusefillc, 2025-04-25:
@FDSoftware @dron0gus does anyone remember if this is now done?

FDSoftware, 2025-04-26:
> [@FDSoftware](https://github.com/FDSoftware) [@dron0gus](https://github.com/dron0gus) does anyone remember if this is now done?

yes!, here: https://github.com/rusefi/rusefi/pull/7701

It would be good if someone tested it before closing it in an ideal world, but from what I could see from the simulator it is working as expected.

closed 2025-04-26 by rusefillc

Abricosvw, 2025-04-26:
it doesn't work for me... i followed the instructions as in this post... the west gate doesn't move at all ... I tried to select TPS, MAP does not work ...

rusefillc, 2025-04-26:
@Abricosvw this ticket is too messy already, please create a new one with your tune attached?