rusefi/rusefi #7151: 36-2-1 trigger error at random times?

ElDominio, 2024-12-10:
**Describe the bug**
I have trigger error *sometimes* on  36-2-1 seems to be when trigger gap ratio is over 2.75, do custom gaps affect this trigger? i have the custom gaps in gap screenshot

**Hardware**
 - Board: custom 4b11 pnp (possible hardware? please help me corroborate)
 - Engine info 4b11

**To Reproduce**
Steps to reproduce the behavior:
1. Idle on a4b11
2. see rpm read errors
3. no errores when over idle speed

**Expected behavior**
no errors

**Screenshots**
log pic:
[image]
gaps:
[image]

**Additional context**
[2024-12-10_18.24.10.mlg.zip](https://github.com/user-attachments/files/18086703/2024-12-10_18.24.10.mlg.zip)

log here

rusefillc, 2024-12-11:
@ElDominio yes custom gaps affect all triggers with sync point

What is the significance of 2.75 for this trigger? Also many previous above 2.75 were fine.

Overall you are doing everything correctly, I do not understand the issue. Best idea I have is we had a quick noise event which logs did not reflect due to how these logs work - the noise is not registered, consecutive legit sync is.

ElDominio, 2024-12-11:
> @ElDominio yes custom gaps affect all triggers with sync point
>
> What is the significance of 2.75 for this trigger? Also many previous above 2.75 were fine.
>
> Overall you are doing everything [...]

I did not know about that on logging, thank you for the info! For real lol

closed 2024-12-11 by ElDominio