rusefi/rusefi #7160: Add Lua hook for Shift Cut Torque Reduction request

ElDominio, 2024-12-14:
**Is your feature request related to a problem? Please describe.**
I would like to be able to set Shift Cut Torque Reduction request

**Would we need new configuration fields?**
Add "Lua" as a request input, basically in the list in the screenshot

**Describe the solution you'd like**
Add an option that says "Lua" that can be set from Lua

**Additional context**
[image]

rusefillc, 2024-12-14:
Somewhat related https://github.com/rusefi/rusefi/issues/7117

kifir23917, 2024-12-15:
@ElDominio, I decided not to add new item to `Activation Mode` combobox and to implement `setTorqueReductionState` Lua hook with approach that is already used for `Clutch Down Switch` and `Brake Pedal`:
`setTorqueReductionState` Lua hook will turn on/off Shift Torque Reduction only when corresponding pin is NONE. If corresponding pin is valid, it will override state configured with `setTorqueReductionState` Lua hook.

closed 2024-12-15 by rusefillc

kifir23917, 2024-12-15:
@ElDominio, `setTorqueReductionState` works only when `Activation Mode` is `Torque Reduction Button` and `Torque Reduction Button` is `NONE`.

[image]

jankovalski, 2024-12-29:
BUG:
If:
`Activation Mode` is `Torque Reduction Button`
and
`Torque Reduction Button` is `NONE`.

Means lua hook is possible but `Settings` are grayed out!
[image]

rusefillc, 2024-12-29:
@jankovalski please create a separate issue

commit e5f0f238 (2024-12-15): Implement `setTorqueReductionState` Lua hook (closes #7160)