rusefi/rusefi #7163: Make IGN CLT Correction a 5x5 table

ElDominio, 2024-12-14:
**Is your feature request related to a problem? Please describe.**
 I'm always frustrated when I want to drop timing at cold idle situations to make the warmup period shorter but the car cannot be driven.

**Would we need new configuration fields?**
Expand the IGN CLT Correction table

**Describe the solution you'd like**
Expand the IGN CLT Correction curve into a 5x5 table

rusefillc, 2024-12-14:
@ElDominio please be more specific / simpler terms

We have

[image]

ElDominio, 2024-12-14:
[image]

rusefillc, 2024-12-14:
👁️ 👁️‍🗨️ 👓 🙄

ElDominio, 2025-01-18:
@FDSoftware any chance you can handle this?

closed 2025-02-20 by rusefillc

ElDominio, 2025-02-20:
[image] not available?

rusefillc, 2025-02-20:
@ElDominio is this not available in https://rusefi.com/build_server/rusefi_bundle_alphax-2chan.zip?
@ElDominio is this not available in https://rusefi.com/build_server/rusefi_bundle_alphax-4chan.zip?
@ElDominio is this not available in https://rusefi.com/build_server/rusefi_bundle_alphax-8chan.zip?
@ElDominio is this not available in _all_ of https://rusefi.com/build_server/?
@ElDominio is this not available in some SPECIFIC of https://rusefi.com/build_server/?

ElDominio, 2025-02-21:
Genesis is broken
H154 does not have it (tested)

5 of my other cars are paralela, does this mean paralela will never have it?

rusefillc, 2025-02-21:
@ElDominio my first attempt to request _exact_ place of so-called `issue` has failed :(

Quck reminder that we have two kinds of Genesis - f4 and f7 are pretty unrelated binaries. 
@ElDominio please spell out which one are you considering to use for this test so that I can manually make sure that that one is fresh?

wider context is that in order to reduce business costs we do not build all binaries on all changes

ElDominio, 2025-02-21:
Car in the shop is h154 f4

Either that or paralela if possible

rusefillc, 2025-02-21:
paralela is nightly so you were acting too soon

@ElDominio I've poked paralela manually. please give it a try

rusefillc, 2025-02-21:
@FDSoftware https://github.com/rusefi/rusefi/commit/a184c1bcda39ce51ccbd19d7545c8adf00703808

ElDominio, 2025-02-24:
When i write -15 on the timing table, 

[image]

it jumps to 10.6, then i set it again to -15 and it sticks, but it's not real:

[image]

rusefillc, 2025-02-24:
@ElDominio smells like something pre-existing with some other correction being added?

ElDominio, 2025-02-24:
The gauges says IGN: CLT Correction is 10.6, while the table is showing my expected value of -15. I will link the tune

ElDominio, 2025-02-24:
[testiattable.zip](https://github.com/user-attachments/files/18950728/testiattable.zip)

i named it iat but is clt

ElDominio, 2025-02-24:
[2025-02-24 18-25-53.zip](https://github.com/user-attachments/files/18950760/2025-02-24.18-25-53.zip)

here is a video of exactly what is happening

tested on both Paralela and H154, same result

FDSoftware, 2025-02-24:
> [@ElDominio](https://github.com/ElDominio) smells like something pre-existing with some other correction being added?

@rusefillc 
updating the ignitionCltCorrTable from int8 with autoscale to float solves the -15 => -10.6 issue
I can't figure out exactly why changing from int8 to float (aka `angle_t`) breaks in this particular case, the IAT corr table seems to do exactly the same thing for quite some time, but I can't find any related issues

@ElDominio With the temperature at `0.0C` the corrections are not executed since it is considered an error (the same thing happens with IAT corrections) This is where the difference between gauge/applied and what the table shows comes from

```c++
angle_t getCltTimingCorrection(float engineLoad) {
	const auto clt = Sensor::get(SensorType::Clt);

	if (!clt)
		return 0; // this error should be already reported somewhere else, let's just handle it
```

rusefillc, 2025-02-25:
@ElDominio @FDSoftware we now have https://github.com/rusefi/rusefi/issues/7501 issue

PR #7475 (2025-02-20): IGN CLT Correction as 5x5 table [was 8*2 curve]
* change curve to table on ign clt correction
* bump FLASH_DATA_VERSION

[image]

resolves #7163