rusefi/rusefi #7169: setBitRangeMsb complaining

ElDominio, 2024-12-14:
https://gist.github.com/ElDominio/0447b1844d7bd5f306263c4cdc435d67

Complaining about something in line 37 being nil

ElDominio, 2024-12-14:
If i use setBitRange NOT msb it does not complain

rusefillc, 2024-12-14:
title says setBitRangeMsb 

https://gist.github.com/ElDominio/0447b1844d7bd5f306263c4cdc435d67 does not contain setBitRangeMsb 

@ElDominio are you invoking a function you did not have a chance to define?

ElDominio, 2024-12-14:
Windows copy paste is broken 167% of the time, i have updated the stupid gist i hate windows

ElDominio, 2024-12-14:
lines 217 and 218 should say "setBitRangeMsb"

rusefillc, 2024-12-14:
@ElDominio please add the following. if does not help please add similar closer to relevant

```
if totalBitIndex == nil then
  print("totalBitIndex is nil")
   return
end
if bitWidth == nil then
  print("bitWidth is nil")
   return
end
```

rusefillc, 2024-12-14:
@ElDominio more protected version of function

```
function setBitRange(data, totalBitIndex, bitWidth, value) 
if totalBitIndex == nil then
  print("totalBitIndex is nil")
   return
end
if bitWidth == nil then
  print("bitWidth is nil")
   return
end
	local byteIndex = totalBitIndex >> 3 
	local bitInByteIndex = totalBitIndex - byteIndex * 8 
	if (bitInByteIndex + bitWidth > 8) then 
		local bitsToHandleNow = 8 - bitInByteIndex 
		setBitRange(data, totalBitIndex + bitsToHandleNow, bitWidth - bitsToHandleNow, value >> bitsToHandleNow) 
		bitWidth = bitsToHandleNow 
	end 
	local mask = (1 << bitWidth) - 1 
local next = data[1 + byteIndex]
if next == nil then
  print("out of range")
   return
end
	data[1 + byteIndex] = next  & (~(mask << bitInByteIndex)) 
	local maskedValue = value & mask 
	local shiftedValue = maskedValue << bitInByteIndex 
	data[1 + byteIndex] = data[1 + byteIndex] | shiftedValue 
end
```

ElDominio, 2024-12-14:
very corrected script updated, thank you

closed 2024-12-14 by ElDominio