rusefi/rusefi #7170: Y label for AFR blends

rusefillc, 2024-12-15:
@ElDominio what should be Y axis for AFR blends? Should it be fuelingLoad like with VE blends or TPSValue like boost blends?

benassi124, 2024-12-15:
It should be able to be either tps or map load

benassi124, 2024-12-15:
Also maybe I am confused but if I for example, want to enrichen lambda target as coolant temperature goes up shouldn't y-axis be coolant temp

ElDominio, 2024-12-15:
Should be fuelingLoad  

@benassi124 , there re two blend tables. Blend table 1 has Ethanol% as param. Blend Table 2 has CLT as param. That was you can add fueling depending on both, independently.

rusefillc, 2024-12-17:
load axis target, load axis blend table and CLT, how are those connected?

[image]

rusefillc, 2024-12-17:
Also

[image]

rusefillc, 2025-02-25:
@FDSoftware is this clear/actionable, or still open complex questions? i know @kifir23917 has made some similar changes about axis naming for other tables

rusefillc, 2025-02-28:
@FDSoftware what's the status here? close or else?

FDSoftware, 2025-02-28:
> [@FDSoftware](https://github.com/FDSoftware) what's the status here? close or else?

we can close it now

rusefillc, 2025-02-28:
@ElDominio please test on platinum-f7 ideally

closed 2025-02-28 by rusefillc