rusefi/rusefi #7182: VE y axis usability issues

rusefillc, 2024-12-17:
* @ElDominio shall we copy "override VE load" to actual VE table dialog?
* attached image should it be showing "TPS %" instead of MAP kpa??

[image]

rusefillc, 2024-12-17:
@ElDominio please approve/review

ElDominio, 2024-12-17:
it SHOULD show TPS on the left side

I don't know if having it duplicated is good, i do not like duplication. I think it should be one of the two, and I like it where it is. VE table should be max focus window for max focus tuning, the other window is seperate. Maybe put some seperators to make it obvious?

kifir23917, 2024-12-27:
@ElDominio, now `VE Table` Y-axis label depends on the value selected in `Override VE table load axis` setting. But this label consists of two parts:

1. main part - can be `load`, `MAP` or `TPS`, depndends on `Override VE table load axis` setting
2. units - can be `kPa`, `MAF`, `%TPS` or `Lua`, depends on `Fuel strategy` setting on `Base Engine Settings` dialog.

As result we can see weird combinations like `TPS kPa`on `VE Table` Y-axis:

[image]

Is it OK?

kifir23917, 2024-12-27:
Also I added `Override VE table load axis` setting on `VE` dialog (see screenshot above). Now it looks consistently with `Boost control open loop' dialog:

[image]

@ElDominio, don't you agree that this `Override VE table load axis` setting looks more obviously on `VE` dialog, than on `Base Engine Settings` dialog?

FDSoftware, 2025-05-02:
> Also I added `Override VE table load axis` setting on `VE` dialog (see screenshot above). Now it looks consistently with `Boost control open loop` dialog:
> 
> [image]
> 
> [@ElDominio](https://github.com/ElDominio), don't you agree that this `Override VE table load axis` setting looks more obviously on `VE` dialog, than on `Injection configuration` dialog?

@ElDominio poke?

ElDominio, 2025-05-03:
I don't like it because it removes focus from main idea of table, which is to tune. When I want to change how table behaves, i like doing it form the general table. I believe this is very personal but I think it makes the table view no longer purely a table view.

When I'm touching tables I like seeing only tables. Dropdown and scalars have their places outside of this

rusefillc, 2025-05-03:
@FDSoftware let me overrule @ElDominio here: for settings which are very related to tables, I believe those settings are allowed to be placed or duplicated next to the table

closed 2025-07-02 by rusefillc