rusefi/rusefi #7195: Add PPS state indicators

rusefillc, 2024-12-20:
We shall improve PPS usability by having a clear "off-throttle now" and "WOT" indicators so that users know that calibration is already good enough

rusefillc, 2024-12-20:
@ElDominio what percentage of PPS precision is totally neglectable? 0.5%? 1%? 3%?

ElDominio, 2024-12-20:
pps is usually not really neglectable since we dpeend on it for target TPS, and especially at low RPM/low load a change of .5% TPS is incredibly noticeable. Precision of 0.1% is required

rusefillc, 2024-12-20:
@ElDominio what about WOT side? do we care much less at WOT?

ElDominio, 2024-12-20:
that is correct, over 85% would be WOT imo

ElDominio, 2024-12-21:
@rusefillc brain actually worked, this is very not important for INDICATION, under 1% PPS should be "off throttle"