rusefi/rusefi #7202: 1KHz logging

rusefillc, 2024-12-21:
Sometimes we want to focus on specific problems, how can we data log 1KHz and more?

all datapoints: X1 rate
only two datapoints: X2 rate?

rusefillc, 2024-12-21:
scatter

rusefillc, 2024-12-21:
https://github.com/rusefi/rusefi/commit/5d5fcf0460832edbf049f17be0486099cde4b74b

rusefillc, 2024-12-21:
ts_protocol.txt
#define TS_GET_SCATTERED_GET_COMMAND '9'

kifir23917, 2024-12-26:
It looks like we could get near 1KHz logging with [11.2.3 Full Optimized - High Speed](https://www.efianalytics.com/TunerStudio/docs/EFI%20Analytics%20ECU%20Definition%20files.pdf) `OutputChannel Capture`  mode, if we limited number `Logged fields`:

[image]

To enable `Full Optimized - High Speed` mode we need to:
- revert commit [484c5be](https://github.com/rusefi/rusefi/commit/484c5bea5ac59316b06591738638821d40e1bbdd) in `rusefi_hd81.ini` file (you can find my file in archive
[rusefi_hd81 (scattered_mode).zip](https://github.com/user-attachments/files/18254156/rusefi_hd81.scattered_mode.zip) and 
- select patched file as `Other / Browse` `Firmware`:
[image]
- select `Slower / Wireless` `Protocol Profile`:
[image]

As result we receive log file with near 1KHz frequency, that can be found in archive [2024-12-26_18.51.44-rpm+uptime (scattered_mode).zip](https://github.com/user-attachments/files/18254226/2024-12-26_18.51.44-rpm%2Buptime.scattered_mode.zip)

rusefillc, 2024-12-26:
@kifir239 please create '1khz' test branch

kifir23917, 2024-12-26:
It looks like we could get near 1KHz logging with [11.2.3 Full Optimized - High Speed](https://www.efianalytics.com/TunerStudio/docs/EFI%20Analytics%20ECU%20Definition%20files.pdf) `OutputChannel Capture`  mode, if we limited number `Logged fields`:

[image]

I've enabled `Full Optimized - High Speed` mode in [test-scattered-get-enabled](https://github.com/rusefi/rusefi/tree/test-scattered-get-enabled) test branch.

closed 2024-12-26 by kifir23917

kifir23917, 2024-12-27:
This functionality was turned off nine months ago because users met some unwanted side-effects because scatter table is located on the same page with the settings.