rusefi/rusefi #7205: Option for divider on NTC temperature inputs

joesphan, 2024-12-22:
**Is your feature request related to a problem? Please describe.**
The current provisions, with a standard 2.49k pullup to 5v resistor, doesn't allow reading of a NTC at lower temperatures without the ADC clipping. For an example my vw clt sensor goes to about 5.9k at 0C. (5.9/(2.4+5.9))*5=3.55V which is over the max adc voltage.
This is the lower of the extremes. Some sensors like the GM CLT will do 9.5k at 0C
[image]

**Would we need new configuration fields?**
ADC divider ratio provisions on NTC thermistor channels, similar to the analog divider ratio under sensors>analog input settings

**Describe the solution you'd like**
Since many sensor and the industry is standard for a 5V pullup, let's provision for a ADC divider in the calculations. For existing builds this can be defaulted to 1, then calculations will be 1:1 to previous firmware. Leaving the field open allows people to run a divider (for ex 5.0V / 2.00 gives a good range of 0v-2.5v).

**Describe alternatives you've considered**
1. using a larger PU resistor. This sacrifices resolution at high temperatures (the temps that matter). Also it decreases support for compatibility across platforms.
2. 3.3V pullup - non industry standard, still requires code changes
3. readjusting calibration values to fit, very clunky user experience.

joesphan, 2024-12-22:
[image]

dron0gus, 2024-12-22:
[image]
Please note R170, R171 resistor arrays.

joesphan, 2024-12-22:
Is mcp's vcc 5v? Looks like the dividers on the output and divide by 2.

dron0gus, 2024-12-22:
Yes, V5A is analog 5V rail.