rusefi/rusefi #7209: transition from batch to sequential skips some fuel pulses

rusefillc, 2024-12-23:
@ggurov reports

[image]

transition from batch to sequential skips some fuel pulses
that's batch inj, sequential ignition
no cam signal, and then cam signal comes back

ggurov, 2024-12-23:
start out as 

batch fuel, sequential igniton - 

no cam signal

[image]

plug cam signal back in

[image]

transitions to sequential ignition, fuel stays batch

transition happens here:

[image]

but there is some fuel skipped when cam signal comes back

[image]

ggurov, 2024-12-23:
[sequential_ign_batch_fuel_transition_to_sequential_broken_1.sal.txt](https://github.com/user-attachments/files/18225375/sequential_ign_batch_fuel_transition_to_sequential_broken_1.sal.txt)

this is .sal

36-2 + 1 tooth cam is the driver

ggurov, 2024-12-23:
above was at about 3500 rpm
this is at cranking:

[image]

ggurov, 2024-12-23:
in simultaneous fuel , there is no missing pulses:

[image]

same simultaneous mode, at the beginning of cranking, i.e. key on -> crank scenario, there's a missing pulse before the priming pulse

this is 250 rpm 

[image]

ggurov, 2024-12-23:
same thing happens  with batch fueling:

[image]

[batch_cranking_still_skips_on_sync.sal.txt](https://github.com/user-attachments/files/18229785/batch_cranking_still_skips_on_sync.sal.txt)

ggurov, 2024-12-23:
oh, i just noticed the missing ignition event in the last two, oh man ,this is a bit more serious.

ggurov, 2024-12-23:
[image]

from total rest.

ggurov, 2024-12-23:
[image]

[jitter_between_cranking.sal.txt](https://github.com/user-attachments/files/18229871/jitter_between_cranking.sal.txt)

when there's jitter between running rpm, as could happen during cold start, or whatever, there are skipped ignition pulses, and skipped fueling events (twice)

joesphan, 2024-12-24:
Here you can hear when it transitions on startup
https://github.com/user-attachments/assets/8eca4ab6-04ce-47c1-877b-b5479afb5f70