rusefi/rusefi #7216: ETB Autotune not working

joesphan, 2024-12-25:
**Describe the bug**
The ETB autotune does not function like it should. I've gotten it to work one time, but every time after that it gets stuck on "start Autotune" phase. The PID values don't update like they should. However they seem to update every time i hit "burn"

**Hardware**
 - Board: Custom F407VGT6
 - TL9201SG
 - corvette C6 pedal
 - Chrysler 75mm ETB

**To Reproduce**
Steps to reproduce the behavior:
1. Advanced>Electronic Throttle Body
2. Auto calibrate ETB1
3. Start ETB PID Autotune
4. PID values only update when you hit burn, TPS is stuck at 20% or so instead of the expected 50%

**Expected behavior**
PID autotune to work

**Screenshots**
https://imgur.com/a/x2wDJBE

rusefillc, 2024-12-29:
@joesphan @Abricosvw see https://github.com/rusefi/rusefi/issues/7152 recent improvement in this area meaning please use latest.

there is also a dream of https://github.com/rusefi/rusefi/issues/7194

joesphan, 2025-01-06:
Not sure what I might have been doing wrong, but i've been driving it around like this. Changing TPS and hitting burn doesn't un-gray them.
[image]
[image]

Although the build is not the latest (20241220) it is after the date of #7152

ggurov, 2025-01-06:
this is the definition of pid autotune button:

```
		commandButton = "Start ETB PID Autotune", cmd_etb_autotune,       {hasIgnitionVoltage && (etbFunctions1 == 1 || etbFunctions1 == 2 || etbFunctions2 == 1 || etbFunctions2 == 2) && calibrationMode = 0}
		commandButton = "Stop ETB PID Autotune",  cmd_etb_autotune_stop,  {hasIgnitionVoltage && (etbFunctions1 == 1 || etbFunctions1 == 2 || etbFunctions2 == 1 || etbFunctions2 == 2) && calibrationMode >= 3 && calibrationMode <= 5}
```

removing hasIgnitionVoltage  from the conditional makes it work again.

so hasIgnitionVoltage isn't getting updated, or it's too low, even though batv reads 12v

rusefillc, 2025-01-06:
What is the firmware here?

Could be related to https://github.com/rusefi/rusefi/issues/7256

joesphan, 2025-01-06:
rusEFI f4_BIGFUEL.2024.12.26.mega144.3912402578

closed 2025-01-06 by rusefillc

joesphan, 2025-01-06:
OK did more testing I have some more insight on this. Seems to be related to my project/tune. Once I recreated the project with minimal settings for ETB to work, it seems to have gone away.
Old tune/project:
[mr2 rusEFI.zip](https://github.com/user-attachments/files/18314373/mr2.rusEFI.zip)

video:
https://youtu.be/lhjmFCSLllo

12:30 is when i recreate a blank project

joesphan, 2025-01-06:
OK lots of insight. So apparently etb pid autotune is highly dependent on feed forward? When i zero out the FF table, it exhibits similar behavior as before. As I add in a bit, it gets closer and closer to 50% before working
[image]
[image]
[image]

commit 0b9e3445 (2025-01-06): ETB Autotune not working fix #7216
commit c71082fc (2025-01-06): ETB Autotune not working fix #7216