rusefi/rusefi #7217: Electronic TB Bias Curve (feed forward) references the wrong Input

joesphan, 2024-12-25:
**Describe the bug**
With the graph open, the X axis references "target TPS Position". I believe this is the intended input, as this will allow setting an open loop ETB curve. However in the chart the label is "TPS", and it actually does reference the actual TPS instead of target TPS. 

**Hardware**
 - Custom F407VGT6
 - v20241107@759225407

**To Reproduce**
Steps to reproduce the behavior:
1. Advanced>Electronic TB Bias Curve

**Expected behavior**
Bias curve to reference target TPS instead of actual TPS

**Screenshots**
[image]

joesphan, 2024-12-26:
Ok so it seems like there was my misunderstanding of feed forward, its meant to adjust the motor such that it stays wherever you move it.
However the X axis is still labelled wrong, should just be TPS and not target TPS position

mck1117, 2024-12-26:
The X axis is not labeled wrong. It's actually using the target, not the current position. The closed loop controller is trying to drive the position to the target anyway, so using the target rather than current position gets you some phase lead and slightly better response.

joesphan, 2024-12-26:
That doesn't seem to be the case, as it moves left/right despite a dashboard gauge with "target tps" open stays at a constant value.

mck1117, 2024-12-26:
> That doesn't seem to be the case, as it moves left/right despite a dashboard gauge with "target tps" open stays at a constant value.

the indication in TS and the actual lookup value in the ECU are unrelated

joesphan, 2025-01-09:
I see now how its working after some testing. it is indeed the target table, the moving indicator just shows how close the current TPS is to the target

closed 2025-01-09 by joesphan