rusefi/rusefi #7223: 2006 Civic Si CAN checksums

ElDominio, 2024-12-28:
https://github.com/ElDominio/CANBUSlogs/blob/main/civic8th/engineon.log

I believe the last byte of id 1DC is some sort of checksum (or key?) required for tacho to agree to show RPM.

If I blast one value for 1DC, it does not show engine RPM. If I blast two consecutive (in terms of time) values for 1DC, I DO get RPM on dash,

Any help is nice

esaulenka, 2025-01-14:
Most likely the upper nibble in the last byte is a counter:
```
1DC 02 00 00 0C
1DC 02 00 00 1B
1DC 02 00 00 2A
1DC 02 00 00 39

1DC 02 00 00 0C
1DC 02 00 00 1B
1DC 02 00 00 2A
1DC 02 00 00 39
....
```
and the lower nibble is a simple sum, calculated by data nibbles.

For `1DC 02 05 9D 2F`
checksum will be 0+2 + 0+5 + 9+0xD + 2+0xF = 0x2E

For `1DC 02 04 B2 38`
checksum is 0+2 + 0+4 + 0xB+2 + 3+8 = 0x1E

So, the final formula is
`chksum = (0x10E - (D0>>4) -  (D0&0xF) - (D1>>4) - (D1&0xF)  - (D2>>4) - (D2&0xF) - (D3>>4) ) & 0xF`

where 0x10E is some magic constant.

But I'm not sure about first byte (D0), it can be not covered be checksum (need more logs to check).

ElDominio, 2025-01-16:
@esaulenka 
I have a ton of logs at https://github.com/ElDominio/CANBUSlogs/tree/main/civic8th

ElDominio, 2025-02-15:
I fed a bunch of info through chatGPT (since I am not very smart about cryptography), and got this
Got it! Since `byte3` consists of a counter (first character) and a checksum (last character), the function should take `byte0`, `byte1`, `byte2`, and the first character of `byte3` as inputs, then calculate the last character.

Based on our analysis, the checksum seems to be:

{checksum nibble} = ({XOR of all bytes} - 1) mod 16

Here's a Lua function that computes the last character of `byte3`:

`function calculate_checksum(byte1, byte2, counter)
    -- byte0 is constant (02), so we include it in XOR calculation
    local byte0 = 0x02
    
    -- Combine known bytes and XOR them
    local xor_result = byte0 ~ byte1 ~ byte2 ~ counter
    
    -- Compute checksum nibble (last character of byte3)
    local checksum_nibble = (xor_result - 1) % 16
    
    -- Convert to hexadecimal character
    return string.format("%X", checksum_nibble)
end`

This Lua function:
- Takes `byte1`, `byte2`, and `counter` as inputs.
- Uses `byte0 = 0x02` (since it's constant).
- Calculates the checksum using XOR and `(XOR - 1) % 16`.
- Converts the result to a hexadecimal character.

I'll be trying this in the coming weeks

ElDominio, 2025-02-20:
https://www.reddit.com/r/CarHacking/comments/tp86pe/comment/i29gdfz/

Also found this, but I am not computer scientist enough to understand what the top comment means.... can you help me @rusefillc ??

rusefillc, 2025-02-20:
@ElDominio voice please call me at your convenience

ElDominio, 2025-04-18:
ChatGPT provided me with actual, tested working stuff
Should it be posted here?

closed 2025-04-18 by ElDominio