rusefi/rusefi #7224: valgrind true or false

rusefillc, 2024-12-28:
* make for valgrind (see bottom of yaml)
* run without suppression https://github.com/rusefi/rusefi/commit/8231adba67128782468b0cc08176723f421e2995

```
==11724== Syscall param write(buf) points to uninitialised byte(s)
==11724==    at 0x4D16574: write (write.c:26)
==11724==    by 0x4C8D964: _IO_file_write@@GLIBC_2.2.5 (fileops.c:1181)
==11724==    by 0x4C8C560: new_do_write (fileops.c:449)
==11724==    by 0x4C8C560: _IO_do_write@@GLIBC_2.2.5 (fileops.c:426)
==11724==    by 0x4C8DA9E: _IO_new_file_xsputn (fileops.c:1244)
==11724==    by 0x4C8DA9E: _IO_file_xsputn@@GLIBC_2.2.5 (fileops.c:1197)
==11724==    by 0x4C80A01: fwrite (iofwrite.c:39)
==11724==    by 0x56F0B1: UnitTestLogBufferWriter::writeInternal(char const*, unsigned long) (unit_test_logger.cpp:12)
==11724==    by 0x56F3A9: BufferedWriter<512>::flush() (buffered_writer.h:55)
==11724==    by 0x56F523: BufferedWriter<512>::write(char const*, unsigned long) (buffered_writer.h:38)
==11724==    by 0x5638AC: writeSdBlock(Writer&) (binary_logging.cpp:126)
==11724==    by 0x56394A: writeSdLogLine(Writer&) (binary_logging.cpp:148)
==11724==    by 0x56F14A: writeUnitTestLogLine() (unit_test_logger.cpp:20)
==11724==    by 0x25B85F: CsvReader::processLine(EngineTestHelper*) (logicdata_csv_reader.cpp:146)
==11724==  Address 0x510dc7b is 1,435 bytes inside a block of size 4,096 alloc'd
==11724==    at 0x4846828: malloc (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==11724==    by 0x4C7F1A4: _IO_file_doallocate (filedoalloc.c:101)
==11724==    by 0x4C8F513: _IO_doallocbuf (genops.c:347)
==11724==    by 0x4C8CF7F: _IO_file_overflow@@GLIBC_2.2.5 (fileops.c:745)
==11724==    by 0x4C8DA9E: _IO_new_file_xsputn (fileops.c:1244)
==11724==    by 0x4C8DA9E: _IO_file_xsputn@@GLIBC_2.2.5 (fileops.c:1197)
==11724==    by 0x4C80A01: fwrite (iofwrite.c:39)
==11724==    by 0x56F0B1: UnitTestLogBufferWriter::writeInternal(char const*, unsigned long) (unit_test_logger.cpp:12)
==11724==    by 0x56F3A9: BufferedWriter<512>::flush() (buffered_writer.h:55)
==11724==    by 0x56F523: BufferedWriter<512>::write(char const*, unsigned long) (buffered_writer.h:38)
==11724==    by 0x563486: LogField::writeHeader(Writer&) const (log_field.cpp:56)
==11724==    by 0x5636C2: writeFileHeader(Writer&) (binary_logging.cpp:82)
==11724==    by 0x563937: writeSdLogLine(Writer&) (binary_logging.cpp:145)
==11724== 
==11724== 
==11724== Exit program on first error (--exit-on-first-error=yes)
```

kifir23917, 2025-01-01:
I located a code that causes the issue: https://github.com/rusefi/rusefi/commit/4620a0a40deba18894fb0a3c887f8115daf5a116#commitcomment-150848151

```
~/rusefi/unit_tests$ make clean
~/rusefi/unit_tests$ make -j8 SANITIZE=no
/rusefi/unit_tests$ valgrind --error-exitcode=1 --exit-on-first-error=yes --leak-check=no --show-error-list=yes --num-callers=500 build/rusefi_test --gtest_filter=arctic.realStartFromFile
```

rusefillc, 2025-01-01:
Passing to @FDSoftware unless @mck1117 would be interested

kifir23917, 2025-01-01:
At list one part of the problem is uninitialized [vvtTargets](https://github.com/search?q=repo%3Arusefi%2Frusefi+vvtTargets&type=code) array elements. But it is not obvious for me where is correct place in the code to initialize these output channels.

P.S. Very stupid experiment allows to increase `fieldIndex` value without grindval failure up to to 300: https://github.com/rusefi/rusefi/commit/079011f45caff655b628c9a63b60e23087cc1feb

rusefillc, 2025-01-01:
@kifir23917 I've improved code generation and that has resolved the issue!

https://github.com/rusefi/rusefi/commit/19178ec2f3655ec6e92e6ab853491ab4aa30e507

closed 2025-01-01 by rusefillc