rusefi/rusefi #7245: technical debt: NT -> US -> NT in unit test scheduler

rusefillc, 2025-01-03:
test_executor.cpp seems to be rounding timeNt while converting to US and back, that's due to historical confusion in unit tests implementation.

Good chance we cannot just switch all tests right away, in that case would be helpful to have a flag for legacy compatible behavior and explicitly set it only in affected tests?

```
  // by the way we have loss of precision while converting NT to integer US
  // technical debt: looks like our unit tests were all created with us precision?
  efitimeus_t timeUs = NT2US(timeNt);
	schedulingQueue.insertTask(scheduling, US2NT(timeUs), action);
```

rusefillc, 2025-01-03:
@FDSoftware another example of test technical debt

FDSoftware, 2025-01-03:
this one is related no? https://github.com/rusefi/rusefi/issues/6457
>until May 2024 unit tests were running with microsecond precision with a defect of comparing microseconds to ticks see https://github.com/rusefi/rusefi/issues/6450

rusefillc, 2025-01-03:
> this one is related no? #6457

"related" probably, which one would have to happen first no idea

FDSoftware, 2025-03-06:
Summary of all tests affected by this change:

```
[  FAILED  ] 59 tests, listed below:
[  FAILED  ] trigger.twoStrokeSingleToothAsSkippedWheel
[  FAILED  ] trigger.twoStrokeSingleToothTrigger
[  FAILED  ] trigger.map_cam_by_magic_point
[  FAILED  ] engine.testRpmAcceleration
[  FAILED  ] engine.testRpmOfCamSensor
[  FAILED  ] engine.testRpmOfTwoStroke
[  FAILED  ] engine.testRpmOfCrankOnly
[  FAILED  ] big.testTriggerInputAdcReal
[  FAILED  ] coil.testOverdwellProtection
[  FAILED  ] miata.miata_na_tdc
[  FAILED  ] arctic.realStartFromFile
[  FAILED  ] realBQS.realHarleyCranking
[  FAILED  ] realBQS.readAsPrimarySensor
[  FAILED  ] realBQS.readAsCam
[  FAILED  ] cranking.realCrankingFromFile
[  FAILED  ] cranking.naCrankFromFile
[  FAILED  ] realCrankingVQ40.normalCrankingSyncCam1
[  FAILED  ] realCrankingVQ40.normalCrankingSyncCam2
[  FAILED  ] real6g72.data1
[  FAILED  ] real6g72.data2
[  FAILED  ] realCas24Plus1.spinningOnBench
[  FAILED  ] real4b11.running
[  FAILED  ] real4b11.runningDoubledEdge
[  FAILED  ] real4g93.cranking
[  FAILED  ] real4g93.crankingCamOnly
[  FAILED  ] fordCoyote.intakeCam
[  FAILED  ] fordCoyote.intakeCamInverted
[  FAILED  ] fordCoyote.exhaustCam
[  FAILED  ] fordCoyote.exhaustCamInverted
[  FAILED  ] crankingVW.vwRealCrankingFromFile
[  FAILED  ] crankingVW.crankingTwiceWithGap
[  FAILED  ] realCrankingNB2.normalCranking
[  FAILED  ] realCrankingNB2.crankingMissingInjector
[  FAILED  ] crankingGm24x_5.gmRealCrankingFromFile
[  FAILED  ] nissan.vq_vvt
[  FAILED  ] nissan.realFromFile
[  FAILED  ] nissan.realNoSparkPlugsFromFile
[  FAILED  ] nissan.realFromFileVVTIN
[  FAILED  ] realk24.crankingNoPlugs1
[  FAILED  ] realk24.crankingNoPlugs2
[  FAILED  ] realk20.cranking
[  FAILED  ] Toyota3ToothCam.RealEngineRunning
[  FAILED  ] RealNoisyTrigger.AvoidOverdwellMinWithInstant
[  FAILED  ] RealNoisyTrigger.AvoidOverdwell1NoInstant
[  FAILED  ] RealNoisyTrigger.AvoidOverdwell1WithInstant
[  FAILED  ] RealNoisyTrigger.AvoidOverdwell2NoInstant
[  FAILED  ] RealNoisyTrigger.AvoidOverdwell2WithInstant
[  FAILED  ] RealNoisyTrigger.AvoidOverdwell3NoInstant
[  FAILED  ] RealNoisyTrigger.AvoidOverdwell3WithInstant
[  FAILED  ] fuelControl.transitionIssue1592
[  FAILED  ] harley.hdCrankingWithCam1
[  FAILED  ] harley.hdCrankingWithCam3
[  FAILED  ] harley.hdCrankingWithCam4
[  FAILED  ] harley.hdCrankingWithCamAnother
[  FAILED  ] harley.hdCrankingWithCamAnother2
[  FAILED  ] harley.hdCrankingWithCamAnother3
[  FAILED  ] ignition.hardwareLatencyCorrection
[  FAILED  ] scheduler.dwellIssue796
[  FAILED  ] HPFP.IntegratedSchedule
```

FDSoftware, 2025-03-06:
I haven't read all 59 tests honestly, but at least the last 5 have in common that they need to emulate RPM, which they do by invoking the following function:

`eth.smartFireTriggerEvents2(/*count*/2, /*delay*/ 20);`

which can be rewritten as:

```cpp
for (int i = 0; i < 2; i++) {
    // aka smartFireRise
    moveTimeForwardAndInvokeEventsUs(MS2US(20));
    eth.firePrimaryTriggerRise();
    // aka smartFireFall
    moveTimeForwardAndInvokeEventsUs(MS2US(20));
    eth.firePrimaryTriggerFall();
}
```

Now, by rewriting it to avoid the MS => US conversion

```cpp
for (int i = 0; i < 2; i++) {
    eth.moveTimeForwardAndInvokeEventsUs(20'000);
    eth.firePrimaryTriggerRise();
    eth.moveTimeForwardAndInvokeEventsUs(20'000);
    eth.firePrimaryTriggerFall();
}
```

we got:
`C++ exception with description "EngineTestHelper: failing to setTimeAndInvokeEventsUs" thrown in the test body.`

Now, if I decide to run over half the logic in `moveTimeForwardAndInvokeEventsUs` it can be simplified as

```cpp
for (int i = 0; i < 2; i++) {
	//eth.moveTimeForwardAndInvokeEventsUs(20'000);
	eth.moveTimeForwardUs(20'000);
	eth.executeActions();
	eth.firePrimaryTriggerRise();

	//eth.moveTimeForwardAndInvokeEventsUs(20'000);
	eth.moveTimeForwardUs(20'000);
	eth.executeActions();
	eth.firePrimaryTriggerFall();
}
```

and finally I have a successful test*

```
[ RUN      ] ignition.hardwareLatencyCorrection
Writing [test_results/unittest_ignition_hardwareLatencyCorrection.msl]
commonInitEngineController
getSpeedDensityAirmass map=96.26
registerInjectionEvent openingAngle=257.59 inj 0
registerInjectionEvent openingAngle=437.59 inj 0
registerInjectionEvent openingAngle=617.59 inj 0
registerInjectionEvent openingAngle=77.59 inj 0
registerInjectionEvent openingAngle=257.59 inj 0
registerInjectionEvent openingAngle=437.59 inj 0
Writing 8 records to test_results/unittest_ignition_hardwareLatencyCorrection.logicdata
Writing 8 records to test_results/unittest_ignition_hardwareLatencyCorrection.events.txt
[       OK ] ignition.hardwareLatencyCorrection (1 ms)
[----------] 1 test from ignition (1 ms total)

[----------] Global test environment tear-down
[==========] 1 test from 1 test suite ran. (2 ms total)
[  PASSED  ] 1 test.
DONE returning 0
All triggers exported to triggers.txt
```

 (taking into account that no event was fired at the appropriate time, as the function mentioned above does)

FDSoftware, 2025-03-06:
more progress:

on `setTimeAndInvokeEventsUs`

we have this call:

```cpp
setTimeNowUs(nextEventTime);
engine.scheduler.executeAll(getTimeNowUs());
```
but executeAll expects NT time no US
`int EventQueue::executeAll(efitick_t now)`
```cpp
setTimeNowUs(nextEventTime);
engine.scheduler.executeAll(getTimeNowNt());
```

after the change the failed test drops to 14:

```
[  FAILED  ] 14 tests, listed below:
[  FAILED  ] coil.testOverdwellProtection
[  FAILED  ] arctic.realStartFromFile
[  FAILED  ] realCas24Plus1.spinningOnBench
[  FAILED  ] real4b11.running
[  FAILED  ] real4b11.runningDoubledEdge
[  FAILED  ] fordCoyote.intakeCamInverted
[  FAILED  ] fordCoyote.exhaustCamInverted
[  FAILED  ] crankingVW.vwRealCrankingFromFile
[  FAILED  ] crankingVW.crankingTwiceWithGap
[  FAILED  ] nissan.vq_vvt
[  FAILED  ] Toyota3ToothCam.RealEngineRunning
[  FAILED  ] fuelControl.transitionIssue1592
[  FAILED  ] ignition.trailingSpark
[  FAILED  ] HPFP.IntegratedSchedule
```

```
Note: Google Test filter = HPFP.IntegratedSchedule
[==========] Running 1 test from 1 test suite.
[----------] Global test environment set-up.
[----------] 1 test from HPFP
[ RUN      ] HPFP.IntegratedSchedule
Writing [test_results/unittest_HPFP_IntegratedSchedule.msl]
commonInitEngineController
getSpeedDensityAirmass map=96.26
registerInjectionEvent openingAngle=280.99 inj 0
registerInjectionEvent openingAngle=460.99 inj 0
registerInjectionEvent openingAngle=640.99 inj 0
registerInjectionEvent openingAngle=100.99 inj 0
registerInjectionEvent openingAngle=280.99 inj 0
registerInjectionEvent openingAngle=460.99 inj 0
registerInjectionEvent openingAngle=640.99 inj 0
registerInjectionEvent openingAngle=280.99 inj 0
registerInjectionEvent openingAngle=460.99 inj 0
registerInjectionEvent openingAngle=640.99 inj 0
registerInjectionEvent openingAngle=100.99 inj 0
registerInjectionEvent openingAngle=143.55 inj 0
registerInjectionEvent openingAngle=323.55 inj 2
registerInjectionEvent openingAngle=503.55 inj 3
registerInjectionEvent openingAngle=683.55 inj 1
registerInjectionEvent openingAngle=143.55 inj 0
registerInjectionEvent openingAngle=323.55 inj 2
registerInjectionEvent openingAngle=503.55 inj 3
tests/test_hpfp_integrated.cpp:46: Failure
Expected equality of these values:
  hpfpTotalToggle
    Which is: 6
  enginePins.hpfpValve.pinToggleCounter
    Which is: 10

Writing 34 records to test_results/unittest_HPFP_IntegratedSchedule.logicdata
Writing 34 records to test_results/unittest_HPFP_IntegratedSchedule.events.txt
[  FAILED  ] HPFP.IntegratedSchedule (4 ms)
```

also i get  a ton of warnings about the coils:
`out-of-order coil off Coil 2unit_test_warning: out-of-order coil off Coil 2`
I think these last ones are the ones that did not fail before due to the NT -> US -> NT conversion.