rusefi/rusefi #7249: HW CI does not handle configuration size reduction right

rusefillc, 2025-01-03:
test change https://github.com/rusefi/rusefi/pull/7248/files which just reduces image size fails HW CI

expected: should not fail HW CI?

```
I 250103 102548.601 [ECU Communication Executor1] BinaryProtocol - pageSize=23244
Error reading current_configuration.binary_image: no worries java.nio.file.NoSuchFileException: current_configuration.binary_image
I 250103 102548.614 [ECU Communication Executor1] BinaryProtocol - Reading from controller...
Error: Exception in thread "ECU Communication Executor1" java.lang.IllegalStateException: TS_RESPONSE_OUT_OF_RANGE ECU/console version mismatch? 22528/716
	at com.rusefi.binaryprotocol.BinaryProtocol.readFullImageFromController(BinaryProtocol.java:362)
	at com.rusefi.binaryprotocol.BinaryProtocol.readImage(BinaryProtocol.java:319)
	at com.rusefi.binaryprotocol.BinaryProtocol.connectAndReadConfiguration(BinaryProtocol.java:189)
	at com.rusefi.io.serial.PortHolder.connectAndReadConfiguration(PortHolder.java:57)
	at com.rusefi.io.serial.StreamConnector.lambda$connectAndReadConfiguration$0(StreamConnector.java:36)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
	at java.base/java.lang.Thread.run(Thread.java:829)
```

rusefillc, 2025-01-03:
@FDSoftware any interest? I know, too many issues :(

rusefillc, 2025-01-03:
for example red master https://github.com/rusefi/rusefi/commit/80c2ee72d4fe1c3773118c6a23afb979c120e64b
follow up commit https://github.com/rusefi/rusefi/commit/4b68967fdb353f950fe94b4586df346a65e72f83

FDSoftware, 2025-01-03:
The HWCI consumes the .ini files to get the signature (and thus rule out if the update failed with a false positive [ex: loading a firmware from a previous run]), in a while I edit the runner and update the ini files before starting with the tests

rusefillc, 2025-01-03:
@FDSoftware says https://github.com/rusefi/rusefi/pull/7251

PR #7252 (2025-01-03): update hardware-ci.yaml for generate config before running tests
resolves #7249
only:config
commit 0df251c7 (2025-01-03): update hardware-ci.yaml for generate config before running tests

resolves #7249
only:config
closed 2025-01-03 by rusefillc