rusefi/rusefi #7253: 3/0 throws trigger error while on bench, using "Enable internal trigger simulation"

ElDominio, 2025-01-04:
**Describe the bug**
Using "Enable internal trigger simulation" sometimes throws trigger erorrs while not being physically in a car

**Hardware**
 - Board: custom f407
 - Engine info 6g72

**To Reproduce**
Steps to reproduce the behavior:
1. set to 3/0
2. enable internal trigger simulation
3. leave and monitor, you will either see trigger errors or VVT angle jumping around

**Expected behavior**
No trigger errors when simulated; could this translate into reality?

**Additional context**
[2025-01-04_15.59.34.mlg.zip](https://github.com/user-attachments/files/18308068/2025-01-04_15.59.34.mlg.zip)

rusefillc, 2025-01-05:
@ElDominio this is pretty far from reality

you have to divide and conquer. You would need to have one ECU producing _external_ signal, then a logic analyzer to record, and then another one to read - that would be a way to divide if you want to troubleshoot simulator which you probably do not care about.

real car too far from all this.

ElDominio, 2025-01-05:
That's pretty much what i was looking for. Should be only affecting simulation and not reality