rusefi/rusefi #7256: Something is not right about EFI_MAIN_RELAY_CONTROL

rusefillc, 2025-01-06:
What is EFI_MAIN_RELAY_CONTROL about? Seems to be set to 'TRUE' on most of the boards with exception of EG33 somehow?!

rusefillc, 2025-01-06:
PS: there is also MRE where it's not defined

rusefillc, 2025-01-06:
Let me at least flip the default to TRUE

FDSoftware, 2025-01-06:
> PS: there is also MRE where it's not defined

found this:

> at the moment microRusEFI would not be happy with vbatt > 5 since microRusEFI senses main relay output

https://github.com/rusefi/rusefi/commit/d97501b6014ee5efc80905d8eef0040cf663821e

also some info about the relay thing:
https://github.com/rusefi/rusefi/pull/1032

> I'm not sure why that means we should turn off the main relay.

> Currently we don't have a lower-power standby mode in firmware. And we don't want to use an ignition switch voltage as a power source for the ECU, because we need some time for proper firmware shutdown and don't want to just instantly kill the firmware with the turn of the key. We may need time to park the stepper motor, or to keep the ECF running some extra seconds to cool down the hot engine, or just wait for engine to stop and save to flash the settings, etc..
> 
> So the idea behind EFI_MAIN_RELAY_CONTROL is to use the main relay output voltage as the power source for ECU instead ignition switch voltage. And then do a self switch-off when the ignition key voltage drops when a user turns the ignition key, but only after the shutdown is complete - that's why we have isInShutdownMode(). So please note that in EFI_MAIN_RELAY_CONTROL we track vBatt as the ignition key voltage, not the ECU power voltage!

mck1117, 2025-01-06:
> So the idea behind EFI_MAIN_RELAY_CONTROL is to use the main relay output voltage as the power source for ECU instead ignition switch voltage.

The other purpose of gating main relay enable on ignition voltage is that some cars have the main relay wired such that the ECU can turn it on without the key on. If the main relay was always enabled while the ECU is on, you'd connect USB (key off) and the main relay would turn on!

rusefillc, 2025-01-06:
> The other purpose of gating main relay enable on ignition voltage is that some cars have the main relay wired such that the ECU can turn it on without the key on. If the main relay was always enabled while the ECU is on, you'd connect USB (key off) and the main relay would turn on!

This part really has to be runtime configuration conditional, not compile time conditional at least on the universal boards?