rusefi/rusefi #7273: lua writePin() to complement readPin()

ggurov, 2025-01-08:
**Is your feature request related to a problem? Please describe.**
there exists a "readPin(pinName)" in lua, but there is no way to actuate an arbitrary pin

**Would we need new configuration fields?**
no

**Describe the solution you'd like**
writePin(pinName) added to lua, such that writePin("PD13", 1) and writePin("PD13", 0) would do the right thing.

**Describe alternatives you've considered**
could just not have programmable inputs/outputs, but would be useful to give generic control in this fashion.

rusefillc, 2025-01-08:
@ggurov I am sorry I am confused since this ticket mentions both readPin and writePin.

Please divide this ticket or just edit to reduce potential confusion?

ggurov, 2025-01-08:
readPin(pinName) exists now, i can do readPin("PD13") and get 1 or 0 bacm.

the ticket is for writePin("PD13", 1), or writePin("PD13", 0).

rusefillc, 2025-01-08:
@ggurov direct access to pins from Lua does not look reasonable, why not use current Lua outputs? See also https://en.wikipedia.org/wiki/XY_problem

ggurov, 2025-01-08:
there are PWM LUA outputs, is the suggestion to use those at 100% dc ?

rusefillc, 2025-01-08:
> there are PWM LUA outputs, is the suggestion to use those at 100% dc ?

yes!