rusefi/rusefi #7280: AFR is richer if TPS is connected

ElDominio, 2025-01-09:
**Describe the bug**
I have a car (Ray's miata) which runs way richer with TPS connected; I can clearly see in logs that there is a difference in injection MS, 6ms when disconnected TPS, 6.7ms when TPS connected.

**Hardware**
 - Board: custom F407
 - Engine info: Miata 1.6
 
**Expected behavior**
AFR to be the same, and injection PW to be the same

**Screenshots**
[image]

**Additional context**
no TPS log data point at 11.25s
yes TPS log data point at 11.28s

miata tune:
[miatatune.zip](https://github.com/user-attachments/files/18367274/miatatune.zip)

[tpslogs.zip](https://github.com/user-attachments/files/18367267/tpslogs.zip)

ElDominio, 2025-01-09:
Fuel: base cycle mass is 4 units higher with TPS than without?

rusefillc, 2025-01-09:
@ElDominio instead of sharing two log files with vague please look at fuel_calculator live data and compare all the intermediate calculation data points there