rusefi/rusefi #7285: Cannot paste in console

ElDominio, 2025-01-11:
Literally cannot CTRL+V in latest console. Right click drop down has no "paste"

ElDominio, 2025-01-11:
[image]

no paste, cannot ctrl v

rusefillc, 2025-01-12:
@ElDominio unable to reproduce, works for me

please try another device, could be unhappy java / clipboard or else?

ElDominio, 2025-01-12:
Does paste show on right click dropdown? Will I have to nuke my PC?

rusefillc, 2025-01-12:
@ElDominio paste does not show in right-click menu but that's very separate from actual ctrl-v which seems to be the focus here?

ElDominio, 2025-01-12:
I only wonder if it is somehow related? Maybe if it was only not showing on mine it was me? In any case I'll try again tomorrow with a different laptop

ElDominio, 2025-01-13:
Script size is the issue, my script is over 8k characters. F427 can't handle that? This is an F4 8chan

rusefillc, 2025-01-13:
Ctrl-v is not broken

@ElDominio please create an issue regarding lua paste usability

closed 2025-01-13 by rusefillc