rusefi/rusefi #7289: SD card un-mount when we software turn off

rusefillc, 2025-01-12:
some boards manage their own power cycle

we need to add SD card un-mount into the turn off procedure

rusefillc, 2025-01-19:
@dron0gus I've just implemented for happy shut-off https://github.com/rusefi/rusefi/commit/14029bc55efdf448a2824ccbacca5b39c351684f

Can we/shall we move into hellenDisableEnSilently which is used by hard error handler?

dron0gus, 2025-01-19:
Strange path for SD card unmount. Why it goes through board-specific code?

No, we should not call such a complicated code from hard fault error handler.

rusefillc, 2025-01-19:
> Strange path for SD card unmount. Why it goes through board-specific code?

yes, but at the moment that's the only code flow where we know that we will be turned off

closed 2025-01-19 by rusefillc