rusefi/rusefi #7291: hasCriticalReportFile method

rusefillc, 2025-01-12:
problem: with progress critical errors would hopefully be less obvious to users #7290

We need a Lua callback to check if SD card contains any critical report files

rusefillc, 2025-01-12:
random fact: @dron0gus plans to upgrade FatFS but that probably does not matter here

rusefillc, 2025-01-17:
@dron0gus could you have removed the unfinished implementation of this under https://github.com/rusefi/rusefi/commit/e0c05c47772642c244370ec4a05cb5bf004ae28d#diff-6942f49238cbf6caf68c40d6edb35f2387ddda0ccd16ae67382778267ead8bc0? "findFile" method was intended for this ticket

rusefillc, 2025-01-18:
@dron0gus skeleton at https://github.com/rusefi/rusefi/commit/65063625ee483a09dc98d9bd9c0c6c95422d0239

dron0gus, 2025-01-22:
Should we simplify "Some error report files exist on SD card" to "We just booted from error fault"?

One old report file will keep annoying user with active indicator.

dron0gus, 2025-01-24:
https://github.com/rusefi/rusefi/pull/7347 and https://github.com/rusefi/rusefi/pull/7349

closed 2025-01-24 by rusefillc