rusefi/rusefi #7300: Access to uninitialized table: tct 20250107@paralela

ElDominio, 2025-01-13:
**Describe the bug**
Title, only happens if I crank with laptop unplugged, plugging in laptop and cranking works fine. sa,e ECU with 20241230 does not have this error, and I have not changed anything in paralela

**Hardware**
 - Board:  paralela 3000GT
 - Engine info 6G72

**To Reproduce**
Steps to reproduce the behavior:
1. Crank with no USB
2. see error led

**Expected behavior**
No error

**Screenshots**
[image]

**Additional context**
[3000gt20250107.zip](https://github.com/user-attachments/files/18402133/3000gt20250107.zip)

rusefillc, 2025-01-14:
probably asking for https://github.com/rusefi/rusefi/commit/3a4a0ba18d9b40060ff7a20ecec76b1195faf6ed#diff-3b1271a72294a43f5015a3b3f3771f2dad70022c082d910d821fe0ca24bd9aac treatment

ElDominio, 2025-01-14:
Is that for today only? Bug is already gone?

commit bcc9a1ff (2025-01-14): Access to uninitialized table: tct fix #7300
commit 9dac54ad (2025-01-14): Access to uninitialized table: tct fix #7300
fix to initialization sequence - noise or starter might get lucky and happen too soon?
only:uaefi
closed 2025-01-14 by rusefillc