rusefi/rusefi #7318: Fan indicators not working on latest fw

ElDominio, 2025-01-17:
**Describe the bug**
Fans do not show, even though fans are activating

**Hardware**
 - Board: custom f427 paralela
 - Engine info K20

**To Reproduce**
Steps to reproduce the behavior:
1. Press fan test button
2. fans turn on, but nothing showing on TS

**Expected behavior**
Fan on button turns on

**Additional context**
Add any other context about the problem here.
[2025-01-16_20.30.38.mlg.zip](https://github.com/user-attachments/files/18446739/2025-01-16_20.30.38.mlg.zip)

rusefillc, 2025-01-17:
@ElDominio thank you! hopefully better now!

ElDominio, 2025-01-17:
Just tried on Platinum using the Test button, no indicator shown and the fan activated

rusefillc, 2025-01-18:
we have two changes here
* code generation was broken while improving things - this part is now fixed
* fan indicators no longer trigger with bench testing. this is because indicators are now implemented closer to fan logic, but bench test is done outside of normal fan control flow :( bench often pokes io lines directly and this is no longer available to live data

commit e3f259a0 (2025-01-17): Fan indicators not working on latest fw fix #7318
closed 2025-01-18 by rusefillc