rusefi/rusefi #7319: Fuel cut code 4: 	FaultRevLimit, What is the fault when no indicators are red?

ElDominio, 2025-01-17:
**Describe the bug**
I am hitting a fuel cut code 4, but I am not seeing what is the fault?

**Hardware**
 - Board: custom f427
 - Engine info k24

**To Reproduce**
Steps to reproduce the behavior:
1. Use latest firmware 20250116
2. rev past about 1800rpm
3. see fuel cut bulb
4. check log
5. reason is 4
6. no more information

**Expected behavior**
No cut code

**Additional context**
Add any other context about the problem here.
[fuelcutcode.zip](https://github.com/user-attachments/files/18446794/fuelcutcode.zip)

Downgrading to 20241031 solved the issue
log with this firmware

[2025-01-16_20.41.42.mlg.zip](https://github.com/user-attachments/files/18446819/2025-01-16_20.41.42.mlg.zip)

rusefillc, 2025-01-17:
@ElDominio quick reminder that documentation exists

step 1: open https://github.com/rusefi/rusefi/wiki/
step 2: in "pages" search control type "codes" or "error"
step 3: https://github.com/rusefi/rusefi/wiki/Error-Codes

looks like 4 is FaultRevLimit?

ElDominio, 2025-01-17:
The title said i searched for "faultrevlimit", literally the title of the thread.

"Faultrevlimit" implies it's a fault causing a rev limit, yet it is not clear what the fault is. Did i not say this in OP?

rusefillc, 2025-01-17:
@ElDominio I've made an idiot out of myself yet another time because I am overwhelmed and fail to handle everything on my plate

short answer there is a 1500 RPM limit in case of ETB fault

longer answer working on making this a bit more obvious now

ElDominio, 2025-01-17:
This limit should be defaulted to enabled, but have a hidden option to disable it under the Experimental tab. Most of my setups are throwing TPS2 Error (and were not doing this before) on latest firmware, which could be the cause of this limit