rusefi/rusefi #7324: Make secondary PPS/TPS requirement less conspicuous

joesphan, 2025-01-19:
**Is your feature request related to a problem? Please describe.**
When you don't set a secondary TPS/PPS, out comes a message saying ETB motor is not configured. I think this needs to be more concise, perhaps "Secondary TPS (PPS) required for ETB to work"

[image]

rusefillc, 2025-01-19:
I agree that current message is not ideal

but it's a non-trivial effort to make this way more useable since a few separate messages would be needed and we have to carefully decide on priority/order of messaging

joesphan, 2025-01-19:
Perhaps you can put a list and I can give a suggestion of priority, especially since I haven't torn down my bench yet