rusefi/rusefi #7325: different update rate of related ign advance corrections

FDSoftware, 2025-01-20:
**Describe the bug**
"IGN CLT Correction" is being updated at a different time than the rest of the corrections

ign clt correcion callstack:
EngineState::periodicFastCallback => updateAdvanceCorrections => getCltTimingCorrection => livedata update all the times

all the other corrections:
periodicFastCallback => getWrappedAdvance => IgnitionState::getAdvance => getAdvanceCorrections => live data update only if we are not in TM_FIXED and there are no engineLoad errors

**Hardware**
 - Board: [any]
 - Engine info [any]

**Expected behavior**
only update livedata of ign clt correction when the rest of the corrections do

**Additional context**

Proposed fix:
`firmware/controllers/algo/ignition/ignition_state.cpp`
```c++
angle_t getAdvanceCorrections(float engineLoad) {
	auto iat = Sensor::get(SensorType::Iat);
	auto clt = Sensor::get(SensorType::Clt);

	if (!iat) {
		engine->ignitionState.timingIatCorrection = 0;
	} else {
		engine->ignitionState.timingIatCorrection = interpolate3d(
			config->ignitionIatCorrTable,
			config->ignitionIatCorrLoadBins, engineLoad,
			config->ignitionIatCorrTempBins, iat.Value
		);
	}

        if(!clt){
              engine->ignitionState.cltTimingCorrection = 0;
          } else {
              engine->ignitionState.cltTimingCorrection = interpolate2d(clt.Value, config->cltTimingBins, config->cltTimingExtra);
        }
....
```

rusefillc, 2025-01-20:
'Bug' is way overblown here so let me abort ticket.

No human no engine would ever notice. For a bug to be legit i think it's required to be observable from outside

closed 2025-01-20 by rusefillc

FDSoftware, 2025-01-20:
> 'Bug' is way overblown here so let me abort ticket.

ok cool, my idea was to send pr with this change to make #7163 easier by already having both corrections (iat/clt) within the same function since both are going to use engineLoad, I'll see to doing it without editing the existing functions then

rusefillc, 2025-01-20:
Just PR as "refactoring/code simplification/something like that", does not need a bug