rusefi/rusefi #7345: WIP ticker: 2GRFE camshaft trigger

ElDominio, 2025-01-22:
**Is your feature request related to a problem? Please describe.**
2/1gr and some newer Toyotas are 4-1 on cam, but with different lengths on each tooth. this trigger doesn't read as rising or falling

**Would we need new configuration fields?**
Dropdown for 2gr

**Describe the solution you'd like**
Add support for 2gr

**Additional context**
[image]

Will have logic analyzer somewhat soon

ElDominio, 2025-01-29:
[goodcamwave.zip](https://github.com/user-attachments/files/18580706/goodcamwave.zip)
[redad.zip](https://github.com/user-attachments/files/18580707/redad.zip)

"redad"  has bad cam scope, but good CKP (standard 36-2)

"goodcamwave" has good cam logic log for 2GRFE 4-1 @rusefillc

ElDominio, 2025-01-29:
[image]

ElDominio, 2025-02-01:
> **Is your feature request related to a problem? Please describe.** 2/1gr and some newer Toyotas are 4-1 on cam, but with different lengths on each tooth. this trigger doesn't read as rising or [...]

Seeing this picture vs oscilloscope makes sense, but even the "goodcamwave" looks strange? Implies 4 teeth? Something wrong with logic analyzer? logic analyzer and oscilloscope were both done on the same car the same day

ElDominio, 2025-03-07:
[image]

mck1117, 2025-03-07:
You can run it as toyota 3 tooth (like a 2jz) on the rising edge and it works fine.

ElDominio, 2025-03-07:
> You can run it as toyota 3 tooth (like a 2jz) on the rising edge and it works fine.

Looking at the cams, I can see it. I'll give it a shot tomorrow and report back, thanks!!!

ElDominio, 2025-04-24:
It does not work as such :S

ElDominio, 2025-04-24:
[2025-04-24_18.04.13.mlg.zip](https://github.com/user-attachments/files/19899791/2025-04-24_18.04.13.mlg.zip)

here is a log with AlphaECU Gold

rusefillc, 2025-04-25:
@ElDominio was shy to admit that he was asking for https://github.com/rusefi/rusefi/commit/8825e7075f0ad8cc14466e0b7279b33d7232ca55

ElDominio, 2025-04-25:
[image]

It seems all three angles are over 100%; refer to log

[image]

rusefillc, 2025-04-25:
expanding range to 255 for now https://github.com/rusefi/rusefi/commit/e530cc642840d5d5a3064380e0f60d1a7258bf06

should do it for even three tooth?

ElDominio, 2025-04-25:
Angle of 

[image]

Has made intake bank 1 vvt very happy, reading at all times

ElDominio, 2025-04-27:
Just tested, we can read all cams with  this setting! precision has to be 50 since vvt swing is about 45 degrees

[image]

closed 2025-04-27 by ElDominio